motioneye-project/motioneye #883: Working scheudle for still images

igrcic, 2018-05-21:
Hello,

firstly, thank you for this very nice Cam software. It is awesome how one can set up and use PiCam in a breeze.

Similar to the issue #766 , I would like to limit my Still Images from dusk to dawn only (taking timelapse 
 pics every few mins from 6am-21pm)

As I see it now, it is not possible as the working schedule settings are only applied to motion detection.

Do you think this would be easy to implement? Im not an python expert, but if you could give me some few tips hints where one should make the changes I think I could come with the patch myself.

Thank you,
Ivan

harmlessdrudge, 2019-11-18:
I'm using MotionEyeOS and have a similar wish (currently camera takes a snapshot every 5 mins). I haven't looked at doing it but wonder if a couple of bash scripts to start and stop a service at set times could be added to crontab? Am just checking here to see if a configuration option has been requested as a feature. Happy to see that it has been.

HOD42, 2021-11-23:
This seems to be a duplicate of https://github.com/ccrisan/motioneyeos/issues/284 too.