motioneye-project/motioneye #885: RTMP Support

clik86, 2018-05-24:
I see previous issues (like #188) on this topic but now motion supports rtmp input stream. I test it with one of my camera. It works on motion alone but the motion eye interface refuse url starting with rtmp scheme.
I look at the code to see how it works but you seam to test the connection before being able to add it and i'm not an RTMP protocol expert so I can't help you.

ccrisan, 2018-05-29:
@clik86 can you give me an example of such an URL?

neur0maniak, 2018-07-20:
I too would like to see this feature implemented.

It should be trivial to set up a VM for testing.
At least in ubuntu, it should be enough to: 
`sudo apt install nginx libnginx-mod-rtmp`
and then add to the bottom of: `/etc/nginx/nginx.conf`
```
rtmp {
        server {
                listen 1935;
                chunk_size 4096;
		notify_method get;
                application live {
                        live on;
                        record off;
                }
        }
}
```
restart nginx `sudo systemctl restart nginx`

send a stream for it to relay:
`ffmpeg -f v4l2 -i /dev/video0 -c:v h264_omx -an -f flv rtmp://xxx.xxx.xxx.xxx/live/mycam`
and the URL to view it will be:
`rtmp://xxx.xxx.xxx.xxx/live/mycam`
where xxx.xxx.xxx.xxx is the ip/fqdn of the nginx server.

epruesse, 2019-02-10:
@ccrisan +1 for me. 

This is helpful for ReoLink cams which suffer from "smearing" when viewed via RTSP. See https://forums.zoneminder.com/viewtopic.php?p=108502#p108502

ReoLink URLs are constructed as follows:

```
rtmp://​{{IP}}​/bcs/channel0_​{{STREAM}}​.bcs?channel=0&stream=0&user=​{{USER}}&password=​{{PASS}}

rtsp://{{USER}}:{{PASS}}​@​{{IP}}​:554//h264Preview_01_{{STREAM}}
```

where `STREAM` is usually `main` (for full res stream). 

Manually editing the `netcam_url` works, so this should be easy. It's supported as of `motion >4.1.0`, although the resolution drop-down disappears and the `width` has to be added manually.

CMDR-Sloma, 2019-04-12:
@ccrisan +1 from me as well

If rtmp works, then this would be perfect as my Reolink cameras in motion eye suffer from low frame rate and smearing. While VLC with rtsp works flawlessly with high resolution and high frame rate motioneye has really sub par performance despite being run on an Intel NUC.

Kanga-Who, 2019-07-19:
Any update on support for the RTMP streams? I have just purchased a Reolink RLC-410-5mp camera and can confirm that in Motioneye smears the RTSP badly, and makes motion capture impossible.

The camera operates flawlessly in VLC, so unsure of the issue.

jasaw, 2019-07-22:
Is the RTMP supported in motion software? Can you guys please run motion without motionEye?

Stop motionEye:
```
systemctl stop motioneye  # for systemd 
/etc/init.d/motioneye stop  # for sysvinit
```

Run motion in the foreground:
```
cd /etc/motioneye
motion -c motion.conf -d 6 -n
```

Check whether RTMP works. If it doesn't, follow [this](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) guide and report it to motion developers.

bwachter, 2019-08-13:
Works with this:

netcam_url rtmp://cam1.cameras/bcs/channel0_sub.bcs?channel=0&stream=1&user=motion&password=

netcam_userpass needed to be empty, I guess the basic auth doesn't work with rtmp.

[-1372602752:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file motion.conf
[-1372602752:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[-1372602752:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
[-1372602752:motion] [NTC] [ALL] motion_startup: Logging to syslog
[-1372602752:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[-1372602752:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[-1372602752:motion] [NTC] [ENC] ffmpeg_global_init: ffmpeg libavcodec version 57.107.100 libavformat version 57.83.100
[0:motion] [NTC] [ALL] main: Camera ID: 1 is from camera-1.conf
[0:motion] [NTC] [ALL] main: Camera ID: 1 Camera Name: Garage Service: rtmp:
[0:motion] [NTC] [ALL] main: Stream port 8081
[0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 16861
[1:ml1:Garage] [NTC] [ALL] motion_init: Camera 1 started: motion detection Enabled
[1:ml1:Garage] [NTC] [VID] vid_start: Opening Netcam RTSP
[0:wc0] [NTC] [STR] http_bindsock: listening on 127.0.0.1 port 7999
[0:wc0] [NTC] [STR] httpd_run: Started motion-httpd server on port 7999 (auth Disabled)
[1:ml1:Garage] [NTC] [NET] netcam_rtsp_connect: Normal resolution: Camera (Garage) connected
[1:ml1:Garage] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[3:nc3:Garage] [NTC] [NET] netcam_rtsp_handler: Normal resolution: Camera handler thread [3] started
[1:ml1:Garage] [NTC] [STR] http_bindsock: listening on any IPv4 address port 8081
[1:ml1:Garage] [NTC] [ALL] motion_init: Started motion-stream server on port 8081 (auth Disabled)
[1:ml1:Garage] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 21 items
[3:nc3:Garage] [NTC] [NET] netcam_rtsp_connect: Normal resolution: Camera (Garage) connected
[1:ml1:Garage] [NTC] [ENC] ffmpeg_set_codec: Low fps. Encoding 4 frames into a 10 frames container.
[1:ml1:Garage] [NTC] [EVT] event_newfile: File of type 8 saved to: /srv/camstor1/Garage/2019-08-13/19-48-25.mp4
[1:ml1:Garage] [NTC] [ALL] motion_detected: Motion detected - starting event 1
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused

thd99, 2019-10-14:
+1, yes please. As mentioned above, with Reolink cameras this should solve smearing issues.

Another example url is: rtmp://x.x.x.x/bcs/channel0_main.bcs?channel=0&stream=0&user=user&password=password.

Haven't tried with motion but works for me with ffmpeg (I'm on 4.1.4-1~deb10u1).

mcd1992, 2019-10-19:
Seems to be working for me as well, need to do more testing though. Should be a simple enough fix to just remove the 'rtmp not supported' code? Will make a PR if so.

1989gironimo, 2021-07-09:
Isn't RTMP supported since several versions? (Or why is this ticket still open?)

starbasessd, 2021-07-09:
It may be that not very many people have reported testing it and it's working. The only ones I've seen have been Reolink users. Hopefully other users of rtmp can report positive or negative...

CMDR-Sloma, 2021-07-19:
I have Reolink RLC-410 5MP and it doesn't work in Motioneye Addon in Home Assistant but the same link works in Frigate which is weird.

starbasessd, 2021-07-19:
Please test as below:
> Is the RTMP supported in motion software? Can you guys please run motion without motionEye?
>
> Stop motionEye:
>
> ```
> systemctl stop motioneye  # for systemd
> /etc/init.d/motioneye stop  # for sysvinit [...]

Also, can you test in a straight motionEyeOS or motionEye on another OS (like Debian or Raspberry Pi OS, instructions here: https://github.com/ccrisan/motioneye/wiki/Installation)
without Home Assistant involved, please? (Use another SD Card or USB stick) 
(motionEyeOS dev20201026 supports USB Boot by changing 1 item in cmdline.txt before first boot: change /dev/mmcblk0p2 to /dev/sda2)

starbasessd, 2021-07-19:
Is the URL you use: 
rtmp://admin:admin@ip_address:1935/bcs/channel0_main.bcs?token=[TOKEN]&channel=0&stream=0
What if you try:
rtmp://ip_address:1935/bcs/channel0_main.bcs?token=[TOKEN]&channel=0&stream=0
and put the username and password in the appropriate boxes?

Marijn0, 2026-08-16:
RTMP has been supported since [v0.42](https://github.com/motioneye-project/motioneye/releases/tag/0.42) in #1508

closed 2026-08-16 by Marijn0