motioneye-project/motioneye #888: port 8081 doesn't stream

cephasiii, 2018-05-29:
I have motioneyeos installed and setup on rpi zero w w/ raspberry camera. When I stream (http://my.ip.address:8081) it only shows a snapshot instead of the live stream. 

How do I view a live video stream without logging in to web interface?

Thanks in advance for the help.
[motion.log](https://github.com/ccrisan/motioneye/files/2046477/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneye/files/2046478/motioneye.log)

francisuk1989, 2018-07-03:
Try this:
Login into the webinterface then go to = on the top left
Go to video streaming > Streaming Port and change 8081 to say 1030
Appy

if this works then there is something on your system that is using port 8081

ikke-t, 2018-07-09:
I noticed the same after upgrade. It seems streaming port is nowadays bound to localhost only. Check if you have 127.0.0.1:8081 in your netstat:

```sudo netstat -ltnp```

If so, you need to find bind setting for motion and put 0.0.0.0 there.

ikke-t, 2018-07-09:
I looked it up and fixed it for me. Option is:

```stream_localhost off```

Works fine. Config option for this would be nice. It's security feature.

ikke-t, 2018-07-09:
Oops, replied to wrong question. Anyhow, it seems streaming is disabled by default nowadays in motion. Sorry for confusion here.

JamBalaya56562, 2026-11-07:
Closing as answered: as found above, Motion serves the stream only to the device itself unless "Video Streaming" is turned on in the camera's settings, which motionEye writes as `stream_localhost off`. With it on, `http://<ip>:8081` streams without logging in to the web UI. This was on motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos).

closed 2026-11-07 by JamBalaya56562