motioneye-project/motioneye #891: Trouble downloading release for raspian

Willsie81, 2018-06-01:
Hi,

Am getting the below error when trying to download to raspian - don't want to use motioneyeos as want to maintain use of the Pi - any advice???

root@raspberrypi:~# wget https://github.com/Motion-Project/motion/releases/downlaod/release-4.1.1/pi_stretch_motion_4.1.1-1_armhf.deb
--2018-06-01 09:41:10--  https://github.com/Motion-Project/motion/releases/downlaod/release-4.1.1/pi_stretch_motion_4.1.1-1_armhf.deb
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... failed: Network is unreachable.
Connecting to github.com (github.com)|192.30.253.112|:443... failed: Network is unreachable.

ccrisan, 2018-06-01:
It looks like a network problem on your side. Definitely not something related to motionEye.

closed 2018-06-01 by ccrisan

nixis, 2018-06-01:
For testing I clicked on the links, both files are not available.

`Not Found`

Seems they are deleted or moved.

Willsie81, 2018-06-01:
I can navigate to these on my laptop and on the pi i can connect to the web etc 

am i missing something? i am new to Pi and this type of thing - had motioneyeos working ok but i want to be able to do more with the pie than it being isolated for just motioneyeOS

Any guidance would be massively appreciated

kleini, 2018-06-01:
This is your link:
https://github.com/Motion-Project/motion/releases/downlaod/release-4.1.1/pi_stretch_motion_4.1.1-1_armhf.deb

And this is the correct link:
https://github.com/Motion-Project/motion/releases/download/release-4.1.1/pi_stretch_motion_4.1.1-1_armhf.deb

I hope, you will see the difference!

Willsie81, 2018-06-01:
Sorry guy's same issue with the correct link
root@raspberrypi:~#  apt-get install ffmpeg v4l-utils
Reading package lists... Done
Building dependency tree
Reading state information... Done
v4l-utils is already the newest version (1.12.3-1).
ffmpeg is already the newest version (7:3.2.10-1~deb9u1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 41 not upgraded.
root@raspberrypi:~# apt-get install libmariadbclient18 libpq5
Reading package lists... Done
Building dependency tree
Reading state information... Done
libmariadbclient18 is already the newest version (10.1.23-9+deb9u1).
libpq5 is already the newest version (9.6.7-0+deb9u1).
0 upgraded, 0 newly installed, 0 to remove and 41 not upgraded.
root@raspberrypi:~# wget https://github.com/Motion-Project/motion/releases/download/release-4.1.1/pi_stretch_motion_4.1.1-1_armhf.deb
--2018-06-01 10:32:44--  https://github.com/Motion-Project/motion/releases/download/release-4.1.1/pi_stretch_motion_4.1.1-1_armhf.deb
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... failed: Network is unreachable.
Connecting to github.com (github.com)|192.30.253.112|:443... failed: Network is unreachable.
root@raspberrypi:~#

kleini, 2018-06-01:
"Network is unreachable" is a clear indicator for wrong network configuration especially regarding the default gateway on your Raspberry Pi. For me the link works perfectly.

Willsie81, 2018-06-01:
Last comment on this - i had attempted to setup a static ip using the Dhcpcd setup and it had disabled my wifi connection for some reason - will be coming back to that in the future - for now all is installing as per instructions - sorry for my lack of knowledge - we've all got to start somewhere