motioneye-project/motioneye #893: Motioneye does not recover from power being cycled

adrian0, 2018-06-03:
Running motioneyeos-raspberrypi-20180401.img.xz
MMAL camera
Raspberry Pi Zero W
WiFi

It seems that if the power is turned off for the raspberry pi running motioneyeos, motioneyeos won't properly boot again once power is turned back on.

After the power is restored, the MotioneyeOS machine shows up when doing a scan of my network. The correct hostname and IP address are seen.
So it seems that something is coming up.

The Web GUI (login screen) never gets loaded, and all other motioneyeos instances on the network are unable to receive the video stream from the 'browned out' camera.

Maybe a brown-out leaves the system in an unsaved or unstable state.

alexandru-2016, 2018-06-24:
Having the same issue with a pi zero w. Will try to get logs next time. Network doesn't start and no output from hdmi.

farid-nasiri, 2019-12-16:
Hi
In raspberry Pi 3 B when  power is lost (without a proper shutdown) and gets back on later , it doesn't show what recorded on that cycle , in web UI it has proper size file but it can't play it. when i downloaded , in windows with kmplayer (with all codec installed) , it doesn't show anything too.

stefansundin, 2021-07-20:
Hello everyone. I just had this problem too on my Raspberry Pi 3B, but I managed to identify why it was happening. My system came up seemingly Ok but it would become unresponsive very quickly.

I had set "Captured Before" in the Motion Detection settings to a high value of 150. Turns out that this caused the RAM on my Pi to fill up with frames, and when the RAM filled up then it attempts to send stuff to the much slower swap file. So everything ground to a halt and I was dumbfounded. I was still able to ping the Pi and `nc -vz 192.168.x.y 22` still showed success. So the kernel was fine but all of userspace was totally unresponsive. I did not have a monitor connected so I don't know if anything showed up there.

After a lot of thinking ("what did I change recently?!"), I finally remembered this setting. So I would recommend everyone to check their settings. It would be nice if motioneye could warn you about dangerous settings too.

I hope this helps someone. :)

closed 2021-07-20 by starbasessd