motioneye-project/motioneye #894: Motioneye settings doesn't show all settings

kahlenberg, 2018-06-04:
Hi,
I have 2 independent system running motioneye. First is Raspberry Pi Zero with camera connected to camera port with ribbon cable. This setup shows every detailed settings like motion threshold, rotate camera image with an angle and so on.
However, second setup doesn't show every settings. This is system consists of BananaPi Pro running Raspbian Jessie, 2 USB webcams are connected to BananaPi over a USB hub. In the settings (port 8756) there few settings. There is no rotating image or motion threshold settings.
What is the problem?

ccrisan, 2018-06-04:
Advanced settings is off, I assume.

kahlenberg, 2018-06-04:
There is no possibility to enable  advanced settings. I have installed motioneye with this: sudo pip install motioneye. I read somewhere that advanced settings is only available in motioneyeOS? But I installed on Raspberry Pi Zero with exactly same command, It doesn't use motioneyeOS, it uses motioneye and it has advanced settings.

ccrisan, 2018-06-04:
Can we see some screenshots? Did you install motion?

kahlenberg, 2018-06-04:
Oh sorry, 
I found it now, it was in front of my nose :). Advanced Settings. Somthings must be diffifult to find in order to find it :)

closed 2018-06-04 by ccrisan