motioneye-project/motioneye #895: ERROR: failed to list mounted disks: 'unmatched'

janpascal, 2018-06-04:
Hi,
I'm using the motioneye package from debian-multimedia, version 0.39 in combination with motion 4.1.1. My motioneye log is contains a lot of lines like this, seemingly two sets after each restart of motioneye:
```
Jun 04 18:14:25 bree meyectl[10172]:    ERROR: failed to list mounted disks: 'unmatched'
Jun 04 18:14:25 bree meyectl[10172]: Traceback (most recent call last):
Jun 04 18:14:25 bree meyectl[10172]:   File "/usr/lib/python2.7/dist-packages/motioneye/diskctl.py", line 222, in list_mounted_disks
Jun 04 18:14:25 bree meyectl[10172]:     disks = _list_disks()
Jun 04 18:14:25 bree meyectl[10172]:   File "/usr/lib/python2.7/dist-packages/motioneye/diskctl.py", line 70, in _list_disks
Jun 04 18:14:25 bree meyectl[10172]:     return _list_disks_dev_by_id()
Jun 04 18:14:25 bree meyectl[10172]:   File "/usr/lib/python2.7/dist-packages/motioneye/diskctl.py", line 134, in _list_disks_dev_by_id
Jun 04 18:14:25 bree meyectl[10172]:     partition.pop('unmatched')
Jun 04 18:14:25 bree meyectl[10172]: KeyError: 'unmatched'
```

Otherwise, motioneye seems to run fine. Does anyone have an idea what could be going on?

Cheers
Jan-Pascal

ccrisan, 2018-06-20:
@janpascal I'll look into it.

ccrisan, 2018-06-24:
Fixed via 5c61470af6ab3fa85dc565ae610b5605535725c2.

closed 2018-06-24 by ccrisan