motioneye-project/motioneye #896: Some issues with MotionEye

THORsHand2018, 2018-06-05:
Hello, 

I have used MotionEye for past 10 months, installed on a 4GB MicroSD card, running on a Raspberry Pi 2B with 2xUSB cameras and 1x TP-Link WiFi camera and here are the issues I encountered
Right now I have the 

OS Version | motionEyeOS 20180224

Trouble with the motion detection, it either starts filming under the mask or doesn't catch the "action" at all, unofrtunally, it hasn't been fully fixed and it has been proven extremely problematic, it either fills up the storage in 1 hr (128GB USB Stick) with useless movies or doesn't film anything, in the version I use right now it's way better but still needs improvement

Trouble with "Movies lifetime": if I set it to erase movies after 3 days, it keeps them and makes the files corrupt, still filling up the USB Stick with them

Trouble with motion notifications, can't get web hooks to work, also email doesn't work

Trouble with reboots, tends to get stuck at network configuration, although it has ethernet connection directly to my switch (Cisco 3500 Series) 

Video streaming is kind of shady, sometimes has good FPS, sometimes it has junk FPS (I've set it to 10 FPS, I have between 1 and 7 FPS)

Tends to make my USB Drive corrupt after a few weeks

Movie upload service barely works, I can't even test SFTP connections, the button does nothing

As a pro, everything that's not mentioned here is a pro, I simply love everything about MotionEye, combined with some DDNS it is the best DYI surveillance system.

electron1979, 2018-12-10:
This repository is for motionEye, NOT motionEyeOS!

JamBalaya56562, 2026-11-08:
Closing, as this was on motionEyeOS 20180224, as noted above, and it bundles several reports with no further replies. motionEyeOS is maintained in its own repository (motioneye-project/motioneyeos). If one of these, such as motion detected under the mask or movies not removed after their lifetime, still happens with a current motionEye, please open a new issue for it.

closed 2026-11-08 by JamBalaya56562