motioneye-project/motioneye #897: Chance to limit file storage maximum

sidealice, 2018-06-05:
If no limitation to file storage of recording. raspberry pi would down without notification and fail to up.

Shall we have chance to config the maximum of the file storage?

darkplatinum, 2018-06-24:
There are some people that create a 100mb partition to prevent this from happening, I'm not sure how to do it so experiment at your own risk.

Ric783, 2021-01-28:
This does seem to be a major flaw in MotionEye. The choice whether to delete the oldest saved files when space runs out seems a very basic feature.

starbasessd, 2021-01-28:
@Ric783 
Issue: Since motionEye is run on other operating systems, with other programs, why would you want to delete video and snapshots from the motionEye files if they were not the cause of the lack of available storage? This would be a 'security' risk, like a buffer overflow, cause file(s) to expand, until the system starts deleting security files... THAT would be a real problem. MotionEyeOS can delete older files (granted, not by space available criteria) automagically, as does motionEye.

Ric783, 2021-01-28:
OK, I am thinking everyone uses MotionEyeOS on a Pi which they only ever look at for a reason, but I don't see why having the option of limiting the specified folder size would be a security risk. Most users are going to be using MotionEye as an alternative to commercial systems, where this is normal. Just writing files until the system stops _is_ a real problem.

starbasessd, 2021-01-28:
You are commenting in the motionEye Issues github, not the motionEyeOS
github.
With OS, there is the simplicity of how long you keep files, vs the
overhead and how often you check to see if the data drive is full...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ric783, 2021-01-28:
My apologies. I did not realise they were different.

starbasessd, 2021-01-28:
No major hu-hu. Different versions for different uses... motionEye is
updated faster and can update whatever piece you want, easier: ffmpeg,
motion, motionEye, python, etc. motionEyeOS is updated/upgraded as a
package, but runs faster/more stable on lesser hardware like SBCs like Pi
and the like...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xerezcd, 2021-02-28:
> If no limitation to file storage of recording. raspberry pi would down without notification and fail to up.
> 
> Shall we have chance to config the maximum of the file storage?

+1 i think it would be nice if they implement this.   (motioneye)

JamBalaya56562, 2026-11-09:
Closing as a duplicate of #302, which asks for the same: removing old files before the disk is full. Please follow and vote there.

closed 2026-11-09 by JamBalaya56562