motioneye-project/motioneye #898: MotionEye fails to start after install - Raspbian Stretch

davesdere, 2018-06-05:
I build motionEye from source then I went to start it but no avail.
I also did the famous
`sudo apt-get update`
and
`sudo apt-get upgrade`

```
sudo systemctl status motioneye

● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-06-05 16:06:58 EDT; 1min 4s ago
 Main PID: 845 (code=exited, status=1/FAILURE)

Jun 05 16:06:58 raspberrypi systemd[1]: Started motionEye Server.
Jun 05 16:06:58 raspberrypi meyectl[845]: Traceback (most recent call last):
Jun 05 16:06:58 raspberrypi meyectl[845]:   File "/usr/local/bin/meyectl", line 7, in <module>
Jun 05 16:06:58 raspberrypi meyectl[845]:     from motioneye.meyectl import main
Jun 05 16:06:58 raspberrypi meyectl[845]:   File "/usr/local/lib/python3.5/dist-packages/motioneye/meyectl.py", line 28, in <module>
Jun 05 16:06:58 raspberrypi meyectl[845]:     import settings
Jun 05 16:06:58 raspberrypi meyectl[845]: ImportError: No module named 'settings'
Jun 05 16:06:58 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Jun 05 16:06:58 raspberrypi systemd[1]: motioneye.service: Unit entered failed state.
Jun 05 16:06:58 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.

```
So I did just to make sure
```
ls /usr/local/lib/python3.5/dist-packages/motioneye/

cleanup.py  handlers.py    meyectl.py     monitor.py    prefs.py     scripts      settings.py  static       templates  uploadservices.py  webhook.py
config.py   __init__.py    mjpgclient.py  motionctl.py  __pycache__  sendmail.py  shell.py     tasks.py     tzctl.py   utils.py           wifictl.py
diskctl.py  mediafiles.py  mmalctl.py     powerctl.py   remote.py    server.py    smbctl.py    template.py  update.py  v4l2ctl.py         wsswitch.py

```
The file is in there. 
I feel it's a simple thing I am missing.
I also tried to install motioneye 4 different ways
```
pip install motioneye
pip3 install motioneye
sudo pip install motioneye
sudo pip3 install motioneye
```
I reloaded the daemon but it still doesn't work. I have no idea why though...

ccrisan, 2018-06-05:
motionEye won't work on python 3.x. Please use python 2.7. What is your default python version (what does `python -V` say)?

davesdere, 2018-06-06:
Thanks for the quick response!
```
python -v

# installing zipimport hook
import zipimport # builtin
# installed zipimport hook
# /usr/lib/python2.7/site.pyc matches /usr/lib/python2.7/site.py
import site # precompiled from /usr/lib/python2.7/site.pyc
# /usr/lib/python2.7/os.pyc matches /usr/lib/python2.7/os.py
import os # precompiled from /usr/lib/python2.7/os.pyc
import errno # builtin
import posix # builtin
# /usr/lib/python2.7/posixpath.pyc matches /usr/lib/python2.7/posixpath.py
import posixpath # precompiled from /usr/lib/python2.7/posixpath.pyc
# /usr/lib/python2.7/stat.pyc matches /usr/lib/python2.7/stat.py
import stat # precompiled from /usr/lib/python2.7/stat.pyc
# /usr/lib/python2.7/genericpath.pyc matches /usr/lib/python2.7/genericpath.py
import genericpath # precompiled from /usr/lib/python2.7/genericpath.pyc
# /usr/lib/python2.7/warnings.pyc matches /usr/lib/python2.7/warnings.py
import warnings # precompiled from /usr/lib/python2.7/warnings.pyc
# /usr/lib/python2.7/linecache.pyc matches /usr/lib/python2.7/linecache.py
import linecache # precompiled from /usr/lib/python2.7/linecache.pyc
# /usr/lib/python2.7/types.pyc matches /usr/lib/python2.7/types.py
import types # precompiled from /usr/lib/python2.7/types.pyc
# /usr/lib/python2.7/UserDict.pyc matches /usr/lib/python2.7/UserDict.py
import UserDict # precompiled from /usr/lib/python2.7/UserDict.pyc
# /usr/lib/python2.7/_abcoll.pyc matches /usr/lib/python2.7/_abcoll.py
import _abcoll # precompiled from /usr/lib/python2.7/_abcoll.pyc
# /usr/lib/python2.7/abc.pyc matches /usr/lib/python2.7/abc.py
import abc # precompiled from /usr/lib/python2.7/abc.pyc
# /usr/lib/python2.7/_weakrefset.pyc matches /usr/lib/python2.7/_weakrefset.py
import _weakrefset # precompiled from /usr/lib/python2.7/_weakrefset.pyc
import _weakref # builtin
# /usr/lib/python2.7/copy_reg.pyc matches /usr/lib/python2.7/copy_reg.py
import copy_reg # precompiled from /usr/lib/python2.7/copy_reg.pyc
# /usr/lib/python2.7/traceback.pyc matches /usr/lib/python2.7/traceback.py
import traceback # precompiled from /usr/lib/python2.7/traceback.pyc
# /usr/lib/python2.7/sysconfig.pyc matches /usr/lib/python2.7/sysconfig.py
import sysconfig # precompiled from /usr/lib/python2.7/sysconfig.pyc
# /usr/lib/python2.7/re.pyc matches /usr/lib/python2.7/re.py
import re # precompiled from /usr/lib/python2.7/re.pyc
# /usr/lib/python2.7/sre_compile.pyc matches /usr/lib/python2.7/sre_compile.py
import sre_compile # precompiled from /usr/lib/python2.7/sre_compile.pyc
import _sre # builtin
# /usr/lib/python2.7/sre_parse.pyc matches /usr/lib/python2.7/sre_parse.py
import sre_parse # precompiled from /usr/lib/python2.7/sre_parse.pyc
# /usr/lib/python2.7/sre_constants.pyc matches /usr/lib/python2.7/sre_constants.py
import sre_constants # precompiled from /usr/lib/python2.7/sre_constants.pyc
import _locale # builtin
# /usr/lib/python2.7/_sysconfigdata.pyc matches /usr/lib/python2.7/_sysconfigdata.py
import _sysconfigdata # precompiled from /usr/lib/python2.7/_sysconfigdata.pyc
# /usr/lib/python2.7/plat-arm-linux-gnueabihf/_sysconfigdata_nd.pyc matches /usr/lib/python2.7/plat-arm-linux-gnueabihf/_sysconfigdata_nd.py
import _sysconfigdata_nd # precompiled from /usr/lib/python2.7/plat-arm-linux-gnueabihf/_sysconfigdata_nd.pyc
# /usr/lib/python2.7/sitecustomize.pyc matches /usr/lib/python2.7/sitecustomize.py
import sitecustomize # precompiled from /usr/lib/python2.7/sitecustomize.pyc
import encodings # directory /usr/lib/python2.7/encodings
# /usr/lib/python2.7/encodings/__init__.pyc matches /usr/lib/python2.7/encodings/__init__.py
import encodings # precompiled from /usr/lib/python2.7/encodings/__init__.pyc
# /usr/lib/python2.7/codecs.pyc matches /usr/lib/python2.7/codecs.py
import codecs # precompiled from /usr/lib/python2.7/codecs.pyc
import _codecs # builtin
# /usr/lib/python2.7/encodings/aliases.pyc matches /usr/lib/python2.7/encodings/aliases.py
import encodings.aliases # precompiled from /usr/lib/python2.7/encodings/aliases.pyc
# /usr/lib/python2.7/encodings/utf_8.pyc matches /usr/lib/python2.7/encodings/utf_8.py
import encodings.utf_8 # precompiled from /usr/lib/python2.7/encodings/utf_8.pyc
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
dlopen("/usr/lib/python2.7/lib-dynload/readline.arm-linux-gnueabihf.so", 2);
[... 33 lines cut ...]
systemctl start motioneye
```
Then I got,
```
root@raspberrypi:~# sudo systemctl status motioneye

● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset
   Active: active (running) since Tue 2018-06-05 22:04:49 EDT; 6s ago
 Main PID: 1779 (meyectl)
   CGroup: /system.slice/motioneye.service
           ├─1779 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/mot
           └─1789 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/mot

Jun 05 22:04:49 raspberrypi systemd[1]: Started motionEye Server.
Jun 05 22:04:49 raspberrypi meyectl[1779]:     INFO: hello! this is motionEye se
Jun 05 22:04:51 raspberrypi meyectl[1779]:     INFO: main config file /etc/motio
Jun 05 22:04:51 raspberrypi meyectl[1779]:     INFO: cleanup started
Jun 05 22:04:51 raspberrypi meyectl[1779]:     INFO: wsswitch started
Jun 05 22:04:51 raspberrypi meyectl[1779]:     INFO: tasks started
Jun 05 22:04:51 raspberrypi meyectl[1779]:     INFO: mjpg client garbage collect
Jun 05 22:04:51 raspberrypi meyectl[1779]:     INFO: server started

```
It's all working now.

closed 2018-06-06 by davesdere

alexlapinski, 2020-02-12:
Has this been fixed? Python 2 is no longer supported.

zagrim, 2020-02-13:
Python 3 support is in progress, see #1208 and #1572