motioneye-project/motioneye #900: motioneye does not start on startup (raspbian)

durahsel, 2018-06-06:
Hi,
I've tried several methods (crontab, autostart with .desktop script ....) but none of them works for me, after boot I've each time to manually start motion eye with this script (which is 'chmod +x')

System

motionEye Version | 0.39
Motion Version | 4.1.1
OS Version | Raspbian 9.4

/home/pi/Desktop/Scripts/motionEyeinitSansSudo.sh

```
#!/bin/bash
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
```

here is my last try with /etc/rc.local method
/etc/rc.local file is 

```
#!/bin/bash
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.
bash /home/pi/Desktop/Scripts/motionEyeinitSansSudo.sh
exit 0
```

also try this one 

```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
# In order to enable or disable this script just change the execution
# bits.
# By default this script does nothing.
sh /home/pi/Desktop/Scripts/motionEyeinitSansSudo.sh
exit 0
```

I know that the script is starting after reboot (trying to add some log file creation at the end of the motionEyeinitSansSudo.sh script and it works) but when I browser motioneye, It does not work. If  I manually start the script after reboot, then it's works

Any idea to make the motionEye works on startup ?

thanks !

electron1979, 2018-12-10:
Did you do step 8: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian?
Did you set 'start_motion_daemon=yes' in '/etc/default/motion'?

JamBalaya56562, 2026-11-08:
Closing, as no startup script is needed any more: `motioneye_init` installs a systemd service and enables it (`systemctl enable motioneye`), so motionEye starts at boot and starts Motion itself. It also turns off Motion's own service, so `start_motion_daemon` in `/etc/default/motion` should stay off. The steps are in the [README](https://github.com/motioneye-project/motioneye#installation).

closed 2026-11-08 by JamBalaya56562