motioneye-project/motioneye #901: RTSP stream losing connection to the camera in a loop

ketansingh, 2018-06-07:
Hi,

I am using `ccrisan/motioneye:dev-armhf` docker image on a Raspberry Pi. From what I understand it comes with its own motion binary. I added my Wyze camera to it using an rtsp stream with the web interface. I enabled debug logs and saw a lot of the following line being repeated:

`netcam_handler_loop: Bad image.  Reconnecting with camera....`

I wanted to store the motion detected video recordings on an external usb drive. I mounted my usb drive at `/mnt/usb` and passed it to the motioneye container using my `docker-compose.yaml`. I set the appropriate mapped path as my Custom Path location via the web UI. I saw the folder being created on disk but videos were not being recorded and saved to the folder. I am sure that motion detection was being triggered because I saw something like this:

`motion_detected: Motion detected - starting event 1`

Then it lost connection to the camera. I also tried storing the videos locally on the Raspberry Pi but it did not help. I tried rebooting the camera as well as Raspberry Pi to no avail. 

Do you guys have any suggestions as to what the problem might be? I would be happy to provide more more information if necessary. I really appreciate your help and thankful for your time.

Thanks!

rainman-xx, 2018-07-31:
Can you enable and post the log?

JamBalaya56562, 2026-11-07:
Closing, as the log asked for above never came. This was with the `dev-armhf` Docker image of 2018. "Bad image. Reconnecting with camera" comes from Motion, which reads the RTSP stream, not from motionEye. If a current image still loses the camera in a loop, please open a new issue with `motion.log`, or report it to [Motion](https://github.com/Motion-Project/motion).

closed 2026-11-07 by JamBalaya56562