motioneye-project/motioneye #902: Raspbian - No USB Cameras

siblight, 2018-06-09:
I have just done a fresh install of Raspbian and Motioneye (using the instructions at https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian).

When i go to add a camera no USB devices are shown.

The devices show in the console:
`pi@raspberrypi:~ $ sudo lsusb
Bus 001 Device 006: ID 046d:080f Logitech, Inc. Webcam C120
Bus 001 Device 005: ID 04b3:310c IBM Corp. Wheel Mouse
Bus 001 Device 004: ID 04b3:3025 IBM Corp. NetVista Full Width Keyboard
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

Any ideas please?

siblight, 2018-06-09:
[image]

Sadly no :(

ccrisan, 2018-06-09:
@siblight please install motion.

JamBalaya56562, 2026-11-07:
Closing, as there was no `/dev/video0`: the system did not create a video device for the webcam, so motionEye had nothing to list. For anyone seeing the same: motionEye lists the devices that `v4l2-ctl --list-devices` shows, so check that command and `dmesg` after plugging the camera in.

closed 2026-11-07 by JamBalaya56562