motioneye-project/motioneye #906: Motion not being detected in motioneye 0.39.1 motion 4.1.1 and Raspbian 9.4 ?

mixpc, 2018-06-17:
Hello,

I have updated from
motionEye Version	0.38.1
Motion Version	4.1
OS Version	Raspbian 9.3

to

motionEye Version	0.39.1
Motion Version	4.1.1
OS Version	Raspbian 9.4

Firstly the OS to 9.4, later on Motioneye via wiki instructions
```
pip install motioneye --upgrade
systemctl restart motioneye
```
 
For some reason, motion detection does not seem to be supported now? Taking a photo every hour, storing in in the could and receiving motificaction works just perfect.

I have previously read related threads:
[https://github.com/ccrisan/motioneye/issues/632](https://github.com/ccrisan/motioneye/issues/632)
[https://github.com/ccrisan/motioneye/issues/848](https://github.com/ccrisan/motioneye/issues/848)

Even a post in German [https://strobelstefan.org/?p=5772](https://strobelstefan.org/?p=5772)
suggesting "daemon on" in /etc/motion/motion.conf

But I have not been able to reach a successful conclusion that allows motion to be detected.

```
pi@raspberrypi:~ $ sudo systemctl status motioneye.service
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-06-17 11:20:19 CEST; 7min ago
 Main PID: 315 (meyectl)
   CGroup: /system.slice/motioneye.service
           ├─315 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           ├─547 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           └─732 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5

Jun 17 11:20:31 raspberrypi meyectl[315]:     INFO: server started
Jun 17 11:26:18 raspberrypi meyectl[315]:  WARNING: 403 GET /login/?_=1529227580820&_username=_&_signature=e3190898ca156a16b25ff88230850c69115fe5cb (1
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [NTC] [ALL] motion_startup: Logging to syslog
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
```

The two lines above are in orange:
_Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
Jun 17 11:27:56 raspberrypi motion[732]: [4355608:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option._

```
pi@raspberrypi:~ $ sudo systemctl status motion.service
● motion.service - LSB: Start Motion detection
   Loaded: loaded (/etc/init.d/motion; generated; vendor preset: enabled)
   Active: active (exited) since Sun 2018-06-17 11:20:19 CEST; 8min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 317 ExecStart=/etc/init.d/motion start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/motion.service

Jun 17 11:20:19 raspberrypi systemd[1]: Starting LSB: Start Motion detection...
Jun 17 11:20:19 raspberrypi motion[317]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Jun 17 11:20:19 raspberrypi systemd[1]: Started LSB: Start Motion detection.
```

It should be noted that motion daemon is not started but I assume that must be ok because of
/usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
since parameter -n tells not to start the daemon

motion.conf from /etc/motion/motion.conf is here [https://pastebin.com/14vCGzNV](https://pastebin.com/14vCGzNV)

motion.conf from /etc/motioneye/motion.conf is here [https://pastebin.com/XTTYyfSX](https://pastebin.com/XTTYyfSX)

motioneye.conf from /etc/motioneye/motion.conf is here [https://pastebin.com/ZjsBm6KZ](https://pastebin.com/ZjsBm6KZ)

thread-1.conf from /etc/motioneye/motion.conf is here [https://pastebin.com/dnN6E6WR](https://pastebin.com/dnN6E6WR)

Any advice on to to get motion detection fixed is much appreciated.

Thank you in advance,

mixpc, 2018-06-17:
It seems this issue has to do with filenaming:

By default:

Image file name %Y-%m-%d-%H-%M-%S #will create a file
Movie file name %Y-%m-%d/%H-%M-%S @#will create a file _within_ a folder

However, when I tried to name movie file just as image file:
%Y-%m-%d-%H-%M-%S
that is, with no folders, jus so that every file is listed cronologically, that is the moment when motion does not seem to work as expected.

@ccrisan could you please confirm that is the expected behaviour?

Any ideas on how to name the video file so that it follows the same reading pattern as the image file?

Thank you in advance,

closed 2018-06-17 by mixpc

ccrisan, 2018-06-18:
@mixpc what you want to achieve with the filenames is perfectly valid from my point of view and should work. The motion log you attached contains only warnings and stuff that should be ignored under normal circumstances.

If you have a specific setup of motion config files (motion.conf + thread-1.conf) that doesn't work as expected due to filename pattern formatting, please attach them here so we can test them ourselves.

Then, if we consider it's a motion problem indeed, we could open an upstream motion issue.