motioneye-project/motioneye #907: Cannot pip install motioneye

hokiejdv, 2018-06-17:
Hi,
I'm using latest stretch lite image on raspberry pi 3b+ and I always get this when trying to install motioneye:
"Collecting motioneye
  Could not find a version that satisfies the requirement motioneye (from versions: )
No matching distribution found for motioneye"
Could you please help me out?
Thank you
Diogo

mixpc, 2018-06-17:
Please, try:
`sudo pip install motioneye`

hokiejdv, 2018-06-17:
If i try this:
sudo pip install git+https://github.com/ccrisan/motioneye
I get this:
Collecting git+https://github.com/ccrisan/motioneye
  Cloning https://github.com/ccrisan/motioneye to /tmp/pip-QbX9Pc-build
Requirement already satisfied: jinja2 in /usr/lib/python2.7/dist-packages (from motioneye==0.39.1)
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (from motioneye==0.39.1)
Collecting pycurl (from motioneye==0.39.1)
  Could not find a version that satisfies the requirement pycurl (from motioneye==0.39.1) (from versions: )
No matching distribution found for pycurl (from motioneye==0.39.1)

hokiejdv, 2018-06-19:
The problem lies with pip version.
One must upgrade to 9.0.3 but not to pip 10 otherwise it does not install motioneye.

ccrisan, 2018-06-19:
@hokiejdv are you sure about that? What are the changes introduced by pip 10 that break motionEye installation?

hokiejdv, 2018-06-21:
Yes, it was a pip problem. I could not install any package with pip 10.
With pip 9.0.1 it was only motioneye I could not install.
After googling around I guess it has something to do with TLS version (1.0 vs 1.2).
After upgrading to pip 10 I've downgraded to pip 9.0.3 and the pip runs smoothly again with no ssl errors.

closed 2018-06-21 by ccrisan