motioneye-project/motioneye #911: No Expert settings in Motioneye installed on  Raspberian / Pi2

DanielGabris, 2018-06-21:
Hello ccrisan, 

I have installed motioneye  via pip on rapsbian (PI2), but I don't see some settings like expert settings so I am not able to turn the raspi camera to FNC. Is that on purpose or have I done something wrong? When installed motioneyeOS, all is fine. I can't use OS as there is no support for my wifi dongle yet and have to install it manualy.
Thanks

Daniel

ccrisan, 2018-06-21:
Those options are only available on motionEyeOS. Sorry about that.

closed 2018-06-21 by ccrisan

zaneclaes, 2020-03-24:
@ccrisan is there an equivalent (fast) solution for Docker + MotionEye on Raspbian base hardware?

ccrisan, 2020-03-25:
@zaneclaes You could try to use https://github.com/ccrisan/streameye yourself.