motioneye-project/motioneye #916: Webpage not loading sometimes

RickvanHek, 2018-06-23:
Hello, sometimes when i connect to my local raspberry motioneye server, it keeps loading with a white screen. I am able to SSH to my raspberry, i can put in the username (admin), it sais "welcome to ...", then after entering the password it is also stuck there.

This seem to  happen arbitrarily, most of the time the first time i connect again after a while. I am using a V4L2 Camera (Trust webcam). Raspberry connected through LAN. I have to manually power off my raspberry and reboot before i can SSH/acces web client again. This is really annoying. If it's in this "state", it also doesn't record anything anymore.

ccrisan, 2018-06-23:
@RickvanHek logs please. It sounds a lot like a network connectivity problem. Also, you're probably on motionEyeOS, not motionEye.

ccrisan, 2018-06-25:
All of them. Everything that you can download via the UI in the logs section.

JamBalaya56562, 2026-11-07:
Closing, as the logs asked for above never came, and this was on motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos). As SSH hung too, the whole system stopped answering, not only motionEye. Please report it there with the logs if it still happens.

closed 2026-11-07 by JamBalaya56562