motioneye-project/motioneye #917: No red frame while recording

biGdada, 2018-06-24:
there is no red frame while recording. is there an on/off switch for this functionality that ive missed?

darkplatinum, 2018-06-24:
It should have a red box. This happens to me once and normal functionality came after a reboot.

biGdada, 2018-06-26:
i know it should. thats exactly why i'm reporting a bug ;)

JamBalaya56562, 2026-11-09:
Sorry this got no answer. The red frame shows when Motion reports the event to motionEye; it is not shown for "Simple MJPEG" cameras. One cause of missing reports, `listen` not including 127.0.0.1, was fixed in #3469 (0.45.0b1). Closing, as there was no news since 2018; if the frame is still missing on a current version, please open a new issue with the motionEye log.