motioneye-project/motioneye #920: configure new camera

davidcgu, 2018-06-25:
Good morning

I have just installed motioneye on my raspberry pi 3 following the steps indicated on the wiki:  https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

I have made the installation under su as was not working directly under pi user.

Now I can open the console and make the logging but when trying to add a camera does not allow me to write nothing on the mandatory box for the camera and don't know what to do to move forward.

[image]

I will appreciate any help.

Thansk in advance

ccrisan, 2018-06-25:
The URL is invalid. Remove the `admin:@` part.

closed 2018-06-25 by ccrisan

davidcgu, 2018-06-25:
wowwwww Thanks a lot ccrisan