motioneye-project/motioneye #921: Question: HTTP API Available

helmar74, 2018-06-25:
Hi,

I've got a question about a HTTP API.
I wrote several scripts (in /data/etc) to control my PTZ camera.
Is there a API existing, where I can run of the scripts out of a Browser (or curl)?
(for a picture snapshot there's an existing URL)

Thanks a a lot

Helmar

nixis, 2018-08-04:
+1 
a API to control motioneye will be a nice feature.

roger-zacharczyk, 2019-01-22:
I think this is related to #972

JamBalaya56562, 2026-11-09:
Scripts set up as action buttons (an executable `<action>_<camera_id>` in the config folder) run with a logged-in `POST /action/<camera_id>/<action>`, which is what the buttons themselves call. A full API is asked for in #972. Closing, as this answers the question.