motioneye-project/motioneye #924: Motion Notifications running when Motion Detection is disabled

Thorinair, 2018-06-27:
Hello!

So, I noticed a rather annoying issue. I wanted to disable motion detection when it is not needed, but all of the Motion Notifications still appear to be executed regardless (and it also appears to be detecting motion in background).

Steps to reproduce:
1. Activate Motion Detection for any camera.
2. Activate a Motion Notification option, like the command one and fill it out with something meaningful so you can track whether it is triggered.
3. Apply the configuration
4. Deactivate Motion Detection for the same camera without touching the Motion Notification options. The Motion Notifications will be hidden as if they are no longer valid.
5. Apply the configuration
6. The Motion Notification will keep firing regardless.

I could also disable the notifications individually, but that deletes the data in the fields, which is equally annoying.

ccrisan, 2018-06-27:
@Thorinair that's very odd since disabling motion detection disables the internal mechanism in the motion daemon. Therefore the events should no longer be generated.

What version of motion do you run?

Thorinair, 2018-06-27:
motionEye Version: 0.39.1
Motion Version: 4.1.1

Thorinair, 2018-06-28:
Update: It appears it even keeps recording movies when the motion detection is disabled. Movies have to completely be disabled as well to prevent it.

ccrisan, 2018-06-28:
Can you please attach your `motion.conf` and your `thread-1.conf` files as they are when your setup behaves as you say?

Thorinair, 2018-06-28:
[motion.zip](https://github.com/ccrisan/motioneye/files/2144160/motion.zip)

Okay, this is with the Command option enabled and set up, applied and then motion detection disabled and applied again. I've put it in zip because GitHub didn't accept the .conf files.

[image]

Thorinair, 2018-06-28:
The reason it was not working was my fault alone... I didn't know exactly what the port defined in `motioneye.conf` did, and I changed it. The result was that it was different from the motion port. Now they are the same and all works. May bad! Closing this

closed 2018-06-28 by Thorinair

nickh66, 2019-05-10:
Hi, 
I was having this issue today myself.  I had simply disabled motion detection while leaving movie\continuous recording enabled.
To stop the emails I found that i had to re-enable motion detection, then disable motion emails, then disable motion detection again.
Is this by design or could there still be a bug?
Cheers
Nick