motioneye-project/motioneye #927: systemctl status motioneye, returns: failed! Result: exit-code*****     Process: 2046 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)

TEfjLONjDON, 2018-06-29:
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2018-06-29 06:37:15 EDT; 11s ago
  Process: 2046 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 2046 (code=exited, status=1/FAILURE)

Jun 29 06:37:15 raspberrypi meyectl[2046]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/diskctl.py", line 22, in <module>
Jun 29 06:37:15 raspberrypi meyectl[2046]:     import utils
Jun 29 06:37:15 raspberrypi meyectl[2046]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/utils.py", line 32, in <module>
Jun 29 06:37:15 raspberrypi meyectl[2046]:     from PIL import Image, ImageDraw
Jun 29 06:37:15 raspberrypi meyectl[2046]:   File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 60, in <module>
Jun 29 06:37:15 raspberrypi meyectl[2046]:     from . import _imaging as core
Jun 29 06:37:15 raspberrypi meyectl[2046]: ImportError: libjpeg.so.8: cannot open shared object file: No such file or directory
Jun 29 06:37:15 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Jun 29 06:37:15 raspberrypi systemd[1]: motioneye.service: Unit entered failed state.
Jun 29 06:37:15 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
~
~
~
~
~
~
~
lines 1-16/16 (END)

Binog, 2018-08-06:
Have the same problem,. how did You recovered from? what is a solution?

JamBalaya56562, 2026-11-08:
Closing, as the traceback shows the cause: it ends with `ImportError: libjpeg.so.8: cannot open shared object file` when Pillow is imported, so the installed Pillow had been built for a libjpeg that the system did not have. This was the Python 2 release; current releases install into a virtual environment with pip, where Pillow comes as a wheel with its own libraries, see the [README](https://github.com/motioneye-project/motioneye#installation).

closed 2026-11-08 by JamBalaya56562