motioneye-project/motioneye #932: Accidentally set my wireless camera unit to the same port as the streaming port

Velkas, 2018-07-04:
I made a stupid decision when trying to simplify my remote access to my makeshift setup. In doing so, I accidentally set my Pi Zero W with camera to use the same network port as the default streaming port (8081). As a matter of saving cpu speed I also disabled all the other network daemons like ssh. Is there any recourse here to access anything other than the streaming url to be able to change the settings again?

francisuk1989, 2018-07-09:
Only way i can think of is taking the SD card and putting this on the computer to access the SD card then go to the etc folder and there would be a motioneye folder and in there should be a confg file of motioneye.conf

Edit the line

> # the TCP port to listen on
> port 8081
to

> # the TCP port to listen on
> port 9955

ccrisan, 2018-07-09:
@Velkas you'll have to log in remotely via putty/SSH and edit your `/data/etc/motioneye.conf` file as @francisuk1989 indicated. Or you can edit the file on the SD card directly, if you run Linux on your PC.

closed 2018-07-09 by ccrisan

Velkas, 2018-07-10:
Thanks for the help guys. Since this remote camera is a Fast Network Camera, the options weren't too annoying to set up again from a fresh flash. It would be cool if the software prevented users from doing silly things like this, but I can understand how that would be low on the docket. I'm back up and running and made sure I kept the ssh daemon running this time 👍