motioneye-project/motioneye #933: Los of video feed

Feesh13, 2018-07-06:
About every 18hrs Motioneye will loose video feed from my 2 cameras even though the cameras are still working properly. Once I reboot Motioneye the video feeds come back and record for about 18hrs.

Motioneye on a RaspberryPi 3
2X Swann NHD-851 cameras

Recording continuous at 1080p 5fps to a 1Tb USB drive that is powered via a separate PSU.

BobMorane31, 2018-07-10:
Hello @Feesh13 

Just a few ideas : 
- what is your USB drive file system (maybe some limits from here) ?
- did you check the "maximum video length" in the "movies" setting of the cameras in motionEye ?

wendy2702, 2018-07-17:
Hi @Feesh13:

have you been able to solve your problem?

It looks that i face similar issue: https://github.com/ccrisan/motioneye/issues/946

Would be nice to get any feedback

Feesh13, 2018-07-17:
@wendy2702 
issue: #946 does look quite similar. 
It still happens. I have resorted to rebooting it daily. I wish I could figure out the issue so I could just let it do its job.

Based on the logs below I'm wondering if the issue may be that the camera is timing out the connection on port 8082 and 8081 and closing it. 

2018-07-11 05:56:26: [motioneye]    ERROR: failed to create movie preview for /data/media/sda1/Laneway/2018-07-04/17-22-22.mp4: Command '[u'ffmpeg', u'-i', u'/data/media/sda1/Laneway/2018-07-04/17-22-22.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/data/media/sda1/Laneway/2018-07-04/17-22-22.mp4.thumb']' returned non-zero exit status 1
2018-07-11 05:56:26: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2018-07-11 05:56:26: [motioneye]  WARNING: mjpg client connection for camera 2 on port 8082 is closed

wendy2702, 2018-07-25:
Hi,

i was able to solve my problem with changing some settings at the streaming client.

Maybe you can check your streaming clients as well.

Br

wendy2702, 2018-07-26:
Yes, but it might be something special. I'm using iobroker to control my Smarthome and have created some  Views which are shown on my wall mounted ipads. 

To view my camera stream i included a "picture widget" to one of my views. This Widget has two additional settings.

1. refresh/reload after wake up
2. refresh/reload after View change

[image]

After i deselect these two options the stream was not interrupted anymore.

JamBalaya56562, 2026-11-09:
In the similar #946 the cause was the viewing client's reload settings. With no news here since 2018, closing. If the feed still stops on a current version, please open a new issue with the motion log from the time it stops.

closed 2026-11-09 by JamBalaya56562