motioneye-project/motioneye #934: How many camera can motioneye on a pc handle?

bobiw, 2018-07-09:
Right now I have 3 raspberry pi running motioneyeos connected to a Linux desktop running motioneye to act as a hub, so I was wondering what is the maximum number of camera can be connected to the hub? and if there is one is it hardware or software related limitation?

ccrisan, 2018-07-15:
It's definitely CPU heavy. RAM consumption really depends on what you do with it, but I'd say no.

bobiw, 2018-07-15:
So a dedicated GPU will not be necessary for a motioneye server/hub correct?

vlad604, 2019-10-25:
Could anyone please comment on specific hardware limit for Raspberry Pi 3 - how many cameras can it practically support. I understand that there is no limit in software, but how many concurrent stream recordings can Raspberry Pi 3 Hub support with reasonable resolution and reasonable fps? Like real life estimates, not just theoretical?