motioneye-project/motioneye #937: Can add camera but can't read rtsp in motionEye

BobMorane31, 2018-07-10:
Hello !
I'm trying to setup a surveillance system at home. Here is my configuration :

-  **PI zero w** : streams video using raspivid & vlc : 
`raspivid -o - -t 0 -w 800 -h 600 -fps 20 -b 180000 | cvlc -vvv stream:///dev/stdin --sout '#rtp{access=udp,sdp=rtsp://:8554/stream}' :demux=h264`

- **Laptop** : Xubuntu + motionEye (for testing, I install motionEye on my server when it works ...)
--  **VLC** successfully reads the rtsp flow from the PI.
-- **motionEye** : I can add the camera (as a network cam) but I only see a grey screen in the motionEye preview with the following message : "**Unable to open video device**".
What seems strange to me is that motionEye gives me the fps received (24.4/20.3 fps) on the bottom left corner ...

Any idea about what is going wrong ??
Thanks :)

### Config
eyeMotion 0.39.2 running on XUbuntu 18.04 (laptop)
Streaming video from RPI zero w with RPI cam V1.3

ccrisan, 2018-07-10:
@BobMorane31 what motion version do you run? Can you provide some logs?

BobMorane31, 2018-07-10:
Hello,
I'm running motion v4.0
I have this problem reported in the log
```
[1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method SETUP failed: 461 Client error
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.103:8554/stream): Protocol not supported
```
But i thought rtsp protocol was supported ... 

Here is the full log from motion eye
```
>> more /var/lib/motioneye/motion.log 

[-725084272:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[-725084272:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
[-725084272:motion] [WRN] [ALL] thread config option deprecated use camera
[-725084272:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[-725084272:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
[-725084272:motion] [NTC] [ALL] motion_startup: Logging to syslog
[-725084272:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[-725084272:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.103:8554/stream)
[1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method SETUP failed: 461 Client error
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.103:8554/stream): Protocol not supported
[1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.103:8554/stream)
[1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method SETUP failed: 461 Client error
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.103:8554/stream): Protocol not supported
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
...

```

ccrisan, 2018-07-11:
@BobMorane31 please upgrade to motion 4.1. If the problem still persists, please [report a motion issue](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue).

BobMorane31, 2018-07-11:
@ccrisan 
I updated motion to V4.1.1, but had no luck with that.
So I opened an issue in motion forum ( [759](https://github.com/Motion-Project/motion/issues/759) ) and will close this one.
Anyway, thanks a lot for taking time to help others on that project :1st_place_medal:

closed 2018-07-11 by BobMorane31