motioneye-project/motioneye #938: pip install was interrupted

Vontux, 2018-07-11:
My installation on a Rasbpian Jessie installation on a Pi3 was interrupted and now whenever I run the following to try and finish the installation so I can follow the steps after 5 on https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian by running `sudo pip install motioneye` I get the following output:

`pi@raspberrypi:~ $ sudo pip install motioneye
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/basecommand.py", line 228, in main
    status = self.run(options, args)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/commands/install.py", line 291, in run
    resolver.resolve(requirement_set)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 103, in resolve
    self._resolve_one(requirement_set, req)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 257, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/resolve.py", line 210, in _get_abstract_dist_for
    self.require_hashes
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/operations/prepare.py", line 310, in prepare_linked_requirement
    progress_bar=self.progress_bar
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/download.py", line 824, in unpack_url
    unpack_file_url(link, location, download_dir, hashes=hashes)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/download.py", line 729, in unpack_file_url
    unpack_file(from_path, location, content_type, link)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/utils/misc.py", line 575, in unpack_file
    flatten=not filename.endswith('.whl')
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/utils/misc.py", line 460, in unzip_file
    zip = zipfile.ZipFile(zipfp, allowZip64=True)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 813, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
BadZipfile: File is not a zip file
`

ccrisan, 2018-07-11:
Please provide a readable output. This is incomprehensible. If you're not familiar with Markdown syntax, please attach it as a file.

Vontux, 2018-07-11:
Better?

ccrisan, 2018-07-11:
A lot better :)

No idea why but pip downloaded an archive that's not really a zip archive. See [this](https://stackoverflow.com/questions/9510474/removing-pips-cache) on how to prevent using the cached archive and redownload the file(s).

Vontux, 2018-07-11:
I actually ended up using a manual installation as I had a time constraint on this particular camera, but in the next day or two I'll be setting up another camera, so I'll try to reproduce the problem and try your solution then, will update soon on whether or not that did the trick.

closed 2022-03-19 by MichaIng