motioneye-project/motioneye #940: Grey screen running rtsp on motioneye os version 4.1.1

kryton79, 2018-07-13:
Hi there. I have recently reinstalled motioneye os. I used this previously and it worked fine with my cameras which are szsinocam P2P IP cameras. I am pretty sure I have set them up the same way as I did before but I am getting a grey screen with the message Unable to open video device. I have my camera set up using rtsp://localipaddress:554/ with the correct username and password but cant seem to be able to see cameras. I'm able to view the live image when logging in to the ip address of the camera directly on my browser so I know its connected and working.

I'm a newblie when it comes to Linux but am able to follow instruction so if anyone can help please point me in the right direction as to where I am going wrong.

davep-uk, 2018-11-24:
Got the same problem, did you fix it ? if so how please.

kryton79, 2018-11-24:
Afraid not..

ruppert92, 2019-07-09:
I have having this issue too. Any resolution?

kryton79, 2019-07-09:
Sorry have given up. Shame as it was a good piece of software.

mcsmark, 2020-12-22:
same problem

starbasessd, 2020-12-22:
Although there are many reasons that casue the gray screen, the current
version of motioneye on another os is 4.3.1.
Since you are not mentioning your base OS install, I will assume it is
Debian - derived, and you can run

pip2 install motioneye --upgrade
systemctl restart motioneye

To see if it picks it up automagically.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-22:
I am using FosCam 99xx rtsp camera with no issues on Ubuntu 20.04 using the
install instructions here:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
and current on Ubuntu Updates.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mcsmark, 2020-12-22:
i use a PI4B with homeassistant. the cam is a network cam. it is recognized but what ever i try it remains gray. 
i also tried other cams. but all the same. dont know if this is a problem of has or motioneye.
i am new to has so maybe i missed something. the stream itself is ok when i used VLC player.

starbasessd, 2020-12-22:
FRENCK is the support for the forked version of motionEye used on
HomeAssistant.
https://community.home-assistant.io/t/home-assistant-community-add-on-motioneye/71826

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-12-22 by starbasessd

mcsmark, 2020-12-22:
thanks, will check there