motioneye-project/motioneye #941: Shutter Speed

UnsterblichMedic, 2018-07-13:
New to this so am not sure where to put this or if this has been addressed.  Love MotionEye btw.

I believe I read that shutter speed was limited to 6 seconds.  I am really looking to do longer shutter speeds for night shots.  I have used the mmalcam_control_params with night, iso, shutter speed but I cannot pull in the night shots I would like.  Is it possible to open up the max shutter speed limit or is there a reason for that?

Using Pi Zero W with Pi Cam V2 and MotionEyeOS

UrasaMinor, 2020-06-26:
I am having the same problem. If you use mmalcam_control_params -a 16, this will turn on the annote feature showing the shutter. You'll notice that the shutter is only ~500,000 ms, or 0.5s. This is because Motioneye has a minimum framerate of 2 fps. Therefore, the longest shutter length it can have the sensor open to take a picture is 0.5s since 1s/0.5 per image = 2 images per second.

So while the camera may allow for shutters up to 6s, Motioneye does not. I am having the same problem. I do not know if Motioneye allows you to decrease the fps below 2, so my apologies there.

JamBalaya56562, 2026-11-13:
As @UrasaMinor found, the limit is the frame rate: Motion does not go below 2 frames per second (with `framerate 1`, Motion 4.7.1 reports `framerate = 2`), so a frame is exposed for at most about 0.5 s. Longer exposures would need Motion itself to allow lower frame rates, so closing here.

closed 2026-11-13 by JamBalaya56562