motioneye-project/motioneye #942: Authentication failed

kduytsch, 2018-07-13:
I'm trying to add a new IP camera (first one was fine), but I'm getting authentical failed.

All parameters are fine as I can get a still image using curl 
 	curl_setopt($curl_handle, CURLOPT_HTTPAUTH, CURLAUTH_DIGEST); 

[image]

And I can't find the log file... I'm on a rasPi v3 running jessie

Thank you

ccrisan, 2018-07-13:
The log can be obtained using `journalctl -u motioneye` or `journalctl -u motioneye > motioneye.log`

kduytsch, 2018-07-13:
This is what I've got, not much info on the reason why it failed

Jul 13 10:24:53 raspberrypi systemd[1]: Started motionEye Server.
Jul 13 10:24:53 raspberrypi meyectl[11245]: WARNING:root:unknown configuration option: setup_mode
Jul 13 10:24:53 raspberrypi meyectl[11245]: WARNING:root:unknown configuration option: webcontrol_parms
Jul 13 10:24:53 raspberrypi meyectl[11245]: WARNING:root:unknown configuration option: webcontrol_localhost
Jul 13 10:24:53 raspberrypi meyectl[11245]: WARNING:root:unknown configuration option: thread
Jul 13 10:24:53 raspberrypi meyectl[11245]: WARNING:root:unknown configuration option: webcontrol_html_output
Jul 13 10:24:53 raspberrypi meyectl[11245]: WARNING:root:unknown configuration option: webcontrol_port
Jul 13 10:24:53 raspberrypi meyectl[11245]:     INFO: hello! this is motionEye server 0.39.2
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "conf_path"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "run_path"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "log_path"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "media_path"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "listen"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "port"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "motion_control_localhost"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "motion_control_port"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "motion_check_interval"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "motion_restart_on_errors"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "mount_check_interval"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "cleanup_interval"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "remote_request_timeout"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "mjpg_client_timeout"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "mjpg_client_idle_timeout"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "smb_shares"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "smb_mount_root"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "enable_reboot"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "smtp_timeout"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "list_media_timeout"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "list_media_timeout_email"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "zip_timeout"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "timelapse_timeout"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "add_remove_cameras"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "http_basic_auth"
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [NTC] [ALL] motion_startup: Logging to syslog
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jul 13 10:24:55 raspberrypi motion[11255]: [10274328:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jul 13 10:24:55 raspberrypi motion[11255]: [1:ml1] [WRN] [NET] netcam_next: called with no data in buffer
Jul 13 10:24:57 raspberrypi meyectl[11245]:     INFO: cleanup started
Jul 13 10:24:57 raspberrypi meyectl[11245]:     INFO: wsswitch started
Jul 13 10:24:57 raspberrypi meyectl[11245]:     INFO: tasks started
Jul 13 10:24:57 raspberrypi meyectl[11245]:     INFO: mjpg client garbage collector started
[... cut ...]
Jul 13 10:25:19 raspberrypi meyectl[11245]: Failed to open /dev/video0: No such file or directory
Jul 13 10:28:35 raspberrypi meyectl[11245]: Failed to open /dev/video0: No such file or directory
Jul 13 10:30:57 raspberrypi meyectl[11245]:    ERROR: failed to list remote cameras on http://192.168.0.31:88/live/ch0: Unknown
Jul 13 10:51:00 raspberrypi motion[11255]: [1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Jul 13 10:57:46 raspberrypi meyectl[11245]:    ERROR: failed to list remote cameras on http://192.168.0.31:88/mjpeg/snap.cgi?chn=1: authentication failed

ccrisan, 2018-07-13:
I think you're using motioneye.conf as a config file for motion somehow. Or you copied it to motion.conf. Or something like that. I suggest that you start with a clean setup, removing all config files.

kduytsch, 2018-07-13:
Done that (I had a link because he couldn't find the motion.conf). Removed now and restarted. Still get the authentication error but can't find logs about it

kduytsch, 2018-07-13:
root@raspberrypi:/etc/motion# journalctl -u motioneye -f |grep meyectl
Jul 13 12:16:00 raspberrypi meyectl[17133]:    DEBUG: creating mjpg client for camera 1
Jul 13 12:16:00 raspberrypi meyectl[17133]:    DEBUG: mjpg client for camera 1 connected on port 8081
Jul 13 12:16:03 raspberrypi meyectl[17133]:    DEBUG: listing cameras
Jul 13 12:16:03 raspberrypi meyectl[17133]:    DEBUG: listing V4L2 devices
Jul 13 12:16:03 raspberrypi meyectl[17133]: Failed to open /dev/video0: No such file or directory
Jul 13 12:16:03 raspberrypi meyectl[17133]:    DEBUG: 200 GET /config/list/?_=1531476962769&username=&password=&proto=v4l2&_username=admin&_signature=c893753ebf658f0a8de9f220586f3a03e34dd39f (192.168.0.25) 23.50ms
Jul 13 12:16:20 raspberrypi meyectl[17133]:    DEBUG: mjpg client for camera 1 on port 8081 has been idle for 10 seconds, removing it
Jul 13 12:16:20 raspberrypi meyectl[17133]:    DEBUG: connection closed for mjpg client for camera 1 on port 8081
Jul 13 12:16:20 raspberrypi meyectl[17133]:    DEBUG: mjpg client for camera 1 on port 8081 removed
Jul 13 12:16:27 raspberrypi meyectl[17133]:    DEBUG: listing cameras
Jul 13 12:16:27 raspberrypi meyectl[17133]:    DEBUG: testing (m)jpg netcam at http://192.168.0.31:88/mjpeg/snap.cgi?chn=1 using basic authentication
Jul 13 12:16:27 raspberrypi meyectl[17133]:    DEBUG: 200 GET /config/list/?_=1531476986688&scheme=http&host=192.168.0.31&port=88&path=%2Fmjpeg%2Fsnap.cgi%3Fchn%3D1&username=xxx&password=xxx&proto=netcam&_username=admin&_signature=a98411c61c58740a3c2f12547244f9725d9cbc10 (192.168.0.25) 89.29ms

Looks like it does basic authentication, but I think I need digest (there was another issue for that with a fix for RSTP)

kduytsch, 2018-07-15:
For now I was able to use RTSP without authentication, less secure, but it allowed me to proceed. It would be great to add the digest auth. On browser, I have that popup authentication that comes on screen.
Thanks

LNFWebsite, 2018-07-23:
Any clues on how to fix this or if this is an unimplemented feature?
Thanks

closed 2018-07-27 by ccrisan

kduytsch, 2018-07-27:
Why did you close it ? rstp allows to fix the problem by removing authentication, but still the problem exists when authentication is set.

ccrisan, 2018-07-27:
@kduytsch I'm lost. So what is the problem here? Are we talking about RTSP or HTTP/MJPG (as your screenshot shows).

kduytsch, 2018-07-27:
Both

When using authentication, none work. For http/MJPG there was no option 
to disable it on my cam. For RTSP, I could and that's why I can now use 
it. But still not securely. If I enable authentication on RTSP, it 
doesn't work.

For record, the authentication is when a prompt comes to screen with 
login/pass (I assume the standard authentication might work but isn't 
supported here)