motioneye-project/motioneye #946: camera stream sporadically lost

wendy2702, 2018-07-16:
Hi,

i installed motioneye a few days ago on my Debian Jessie Homeserver. Installation went finde without any issues but now i faced a problem that my rtsp stream seems to stop working as i did not get any camera picture anymore from motioneye WEB-IF or any client.

Installed Motioneye Version: 0.39.2
Motion: 4.1.1
Debian: 8.11 (up to date)

System Disk Kingston SSD 120GB with 70GB free space
Storage Device: Samsung 500GB WDU 75% usage
Camera: Hikvision DS-2CD2042WD-I

Attached you can find a logfile. Problem starts at:

`Jul 16 17:10:53 openmediavault meyectl[12137]: ERROR: mjpg client timed out receiving data for camera 1 on port 8085`

Maybe somebody can tell me if this is a setup issue or how to fix. Thansk for help

```Jul 16 17:04:13 openmediavault meyectl[12137]: DEBUG: adding task "make_movie_preview(/srv/dev-disk-by-label-Cameras/2018-07-16/17-02-11.mp4)" in 5 seconds
Jul 16 17:04:13 openmediavault meyectl[12137]: DEBUG: saving tasks to "/etc/motioneye/tasks.pickle"...
Jul 16 17:04:13 openmediavault meyectl[12137]: DEBUG: 200 POST /_relay_event/?_username=admin&event=movie_end&thread_id=1&_signature=dcf1f047486846057cbeb5a57511cdf49dd2d74e (127.0.0.1) 1.02ms
Jul 16 17:04:13 openmediavault motion: [1:ml1] [NTC] [EVT] event_newfile: File of type 8 saved to: /srv/dev-disk-by-label-Cameras/2018-07-16/17-04-11.mp4
Jul 16 17:04:13 openmediavault motion: [1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 879
Jul 16 17:04:13 openmediavault meyectl[12137]: DEBUG: received relayed event start for thread id 1 (camera id 1)
Jul 16 17:04:13 openmediavault meyectl[12137]: DEBUG: ignoring start event for camera with id 1 and motion detection disabled
Jul 16 17:04:13 openmediavault meyectl[12137]: DEBUG: 200 POST /_relay_event/?_username=admin&event=start&thread_id=1&_signature=eca3cdc2b0194c21ec09b507c8db91fbd1c14937 (127.0.0.1) 0.63ms
Jul 16 17:04:14 openmediavault motion: [1:ml1] [INF] [ENC] ffmpeg_avcodec_log: specified frame type (3) at 0 is not compatible with keyframe interval
Jul 16 17:04:18 openmediavault meyectl[12137]: DEBUG: executing task "make_movie_preview(/srv/dev-disk-by-label-Cameras/2018-07-16/17-02-11.mp4)"
Jul 16 17:04:18 openmediavault meyectl[12137]: DEBUG: saving tasks to "/etc/motioneye/tasks.pickle"...
Jul 16 17:04:18 openmediavault meyectl[12137]: DEBUG: creating movie preview for /srv/dev-disk-by-label-Cameras/2018-07-16/17-02-11.mp4 with an offset of 4 seconds...
Jul 16 17:04:18 openmediavault meyectl[12137]: DEBUG: running command "ffmpeg -i /srv/dev-disk-by-label-Cameras/2018-07-16/17-02-11.mp4 -f mjpeg -vframes 1 -ss 4 -y /srv/dev-disk-by-label-Cameras/2018-07-16/17-02-11.mp4.thumb"
Jul 16 17:06:13 openmediavault motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop 1'
Jul 16 17:06:13 openmediavault motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end 1 /srv/dev-disk-by-label-Cameras/2018-07-16/17-04-11.mp4'
Jul 16 17:06:13 openmediavault motion: [1:ml1] [NTC] [ALL] mlp_actions: End of event 879
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: received relayed event stop for thread id 1 (camera id 1)
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: clearing motion detected for camera with id 1
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=50a22f432f6c34897d0f5a6129d691051629c20a (127.0.0.1) 0.76ms
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: received relayed event movie_end for thread id 1 (camera id 1)
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: adding task "make_movie_preview(/srv/dev-disk-by-label-Cameras/2018-07-16/17-04-11.mp4)" in 5 seconds
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: saving tasks to "/etc/motioneye/tasks.pickle"...
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: 200 POST /_relay_event/?_username=admin&event=movie_end&thread_id=1&_signature=b75f46399f545ad9e48f5fc3a52fc368ae9b8615 (127.0.0.1) 0.90ms
Jul 16 17:06:13 openmediavault motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 1'
Jul 16 17:06:13 openmediavault motion: [1:ml1] [INF] [EVT] event_new_video: Source FPS 15
Jul 16 17:06:13 openmediavault motion: [1:ml1] [WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted
Jul 16 17:06:13 openmediavault motion: [1:ml1] [NTC] [ENC] ffmpeg_set_codec: Using codec libx264
Jul 16 17:06:13 openmediavault motion: [1:ml1] [INF] [ENC] ffmpeg_set_quality: libx264 codec vbr/crf/bit_rate: 12
Jul 16 17:06:13 openmediavault motion: [1:ml1] [NTC] [EVT] event_newfile: File of type 8 saved to: /srv/dev-disk-by-label-Cameras/2018-07-16/17-06-11.mp4
Jul 16 17:06:13 openmediavault motion: [1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 880
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: received relayed event start for thread id 1 (camera id 1)
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: ignoring start event for camera with id 1 and motion detection disabled
Jul 16 17:06:13 openmediavault meyectl[12137]: DEBUG: 200 POST /_relay_event/?_username=admin&event=start&thread_id=1&_signature=eca3cdc2b0194c21ec09b507c8db91fbd1c14937 (127.0.0.1) 1.20ms
Jul 16 17:06:14 openmediavault motion: [1:ml1] [INF] [ENC] ffmpeg_avcodec_log: specified frame type (3) at 0 is not compatible with keyframe interval
Jul 16 17:06:18 openmediavault meyectl[12137]: DEBUG: executing task "make_movie_preview(/srv/dev-disk-by-label-Cameras/2018-07-16/17-04-11.mp4)"
Jul 16 17:06:18 openmediavault meyectl[12137]: DEBUG: saving tasks to "/etc/motioneye/tasks.pickle"...
Jul 16 17:06:18 openmediavault meyectl[12137]: DEBUG: creating movie preview for /srv/dev-disk-by-label-Cameras/2018-07-16/17-04-11.mp4 with an offset of 4 seconds...
Jul 16 17:06:18 openmediavault meyectl[12137]: DEBUG: running command "ffmpeg -i /srv/dev-disk-by-label-Cameras/2018-07-16/17-04-11.mp4 -f mjpeg -vframes 1 -ss 4 -y /srv/dev-disk-by-label-Cameras/2018-07-16/17-04-11.mp4.thumb"
Jul 16 17:07:11 openmediavault motion: [1:ml1] [INF] [ENC] ffmpeg_avcodec_log: specified frame type (3) at 851 is not compatible with keyframe interval
Jul 16 17:08:13 openmediavault motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop 1'
Jul 16 17:08:13 openmediavault motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end 1 /srv/dev-disk-by-label-Cameras/2018-07-16/17-06-11.mp4'
Jul 16 17:08:13 openmediavault motion: [1:ml1] [NTC] [ALL] mlp_actions: End of event 880
Jul 16 17:08:13 openmediavault motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 1'
Jul 16 17:08:13 openmediavault motion: [1:ml1] [INF] [EVT] event_new_video: Source FPS 15
Jul 16 17:08:13 openmediavault motion: [1:ml1] [WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted
Jul 16 17:08:13 openmediavault motion: [1:ml1] [NTC] [ENC] ffmpeg_set_codec: Using codec libx264
Jul 16 17:08:13 openmediavault motion: [1:ml1] [INF] [ENC] ffmpeg_set_quality: libx264 codec vbr/crf/bit_rate: 12
Jul 16 17:08:13 openmediavault meyectl[12137]: DEBUG: received relayed event stop for thread id 1 (camera id 1)
Jul 16 17:08:13 openmediavault meyectl[12137]: DEBUG: clearing motion detected for camera with id 1
Jul 16 17:08:13 openmediavault meyectl[12137]: DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=50a22f432f6c34897d0f5a6129d691051629c20a (127.0.0.1) 0.64ms
Jul 16 17:08:13 openmediavault meyectl[12137]: DEBUG: received relayed event movie_end for thread id 1 (camera id 1)
Jul 16 17:08:13 openmediavault meyectl[12137]: DEBUG: adding task "make_movie_preview(/srv/dev-disk-by-label-Cameras/2018-07-16/17-06-11.mp4)" in 5 seconds
[... 121 lines cut ...]
[... cut ...]
Jul 16 17:14:23 openmediavault meyectl[12137]: ERROR: mjpg client timed out receiving data for camera 1 on port 8085
Jul 16 17:14:33 openmediavault meyectl[12137]: ERROR: mjpg client timed out receiving data for camera 1 on port 8085
Jul 16 17:14:43 openmediavault meyectl[12137]: ERROR: mjpg client timed out receiving data for camera 1 on port 8085
Jul 16 17:14:53 openmediavault meyectl[12137]: ERROR: mjpg client timed out receiving data for camera 1 on port 8085
Jul 16 17:15:01 openmediavault rrdcached[1472]: Received FLUSHALL`

wendy2702, 2018-07-17:
Few minutes ago stream stopped again.

This is from motion log after changing any value at WEB-IF.

```[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
[0:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (ALL)
[0:motion] [INF] [ALL] conf_output_parms: Writing configuration parameters from all files (2):
[0:motion] [INF] [ALL] Thread 0 - Config file: /etc/motioneye/motion.conf
[0:motion] [INF] [ALL] daemon                    off
[0:motion] [INF] [ALL] setup_mode                off
[0:motion] [INF] [ALL] log_level                 9
[0:motion] [INF] [ALL] log_type                  ALL
[0:motion] [INF] [ALL] videodevice               /dev/video0
[0:motion] [INF] [ALL] v4l2_palette              17
[0:motion] [INF] [ALL] input                     -1
[0:motion] [INF] [ALL] norm                      0
[0:motion] [INF] [ALL] frequency                 0
[0:motion] [INF] [ALL] flip_axis                 none
[0:motion] [INF] [ALL] rotate                    0
[0:motion] [INF] [ALL] width                     352
[0:motion] [INF] [ALL] height                    288
[0:motion] [INF] [ALL] framerate                 100
[0:motion] [INF] [ALL] minimum_frame_time        0
[0:motion] [INF] [ALL] netcam_keepalive          off
[0:motion] [INF] [ALL] netcam_tolerant_check     off
[0:motion] [INF] [ALL] rtsp_uses_tcp             on
[0:motion] [INF] [ALL] auto_brightness           off
[0:motion] [INF] [ALL] brightness                0
[0:motion] [INF] [ALL] contrast                  0
[0:motion] [INF] [ALL] saturation                0
[0:motion] [INF] [ALL] hue                       0
[0:motion] [INF] [ALL] power_line_frequency      -1
[0:motion] [INF] [ALL] roundrobin_frames         1
[0:motion] [INF] [ALL] roundrobin_skip           1
[0:motion] [INF] [ALL] switchfilter              off
[0:motion] [INF] [ALL] threshold                 1500
[0:motion] [INF] [ALL] threshold_tune            off
[0:motion] [INF] [ALL] noise_level               32
[0:motion] [INF] [ALL] noise_tune                on
[0:motion] [INF] [ALL] smart_mask_speed          0
[0:motion] [INF] [ALL] lightswitch               0
[0:motion] [INF] [ALL] minimum_motion_frames     1
[0:motion] [INF] [ALL] pre_capture               0
[0:motion] [INF] [ALL] post_capture              0
[0:motion] [INF] [ALL] event_gap                 60
[0:motion] [INF] [ALL] max_movie_time            3600
[0:motion] [INF] [ALL] emulate_motion            off
[0:motion] [INF] [ALL] output_pictures           on
[0:motion] [INF] [ALL] output_debug_pictures     off
[0:motion] [INF] [ALL] quality                   75
[0:motion] [INF] [ALL] camera_id                 0
[0:motion] [INF] [ALL] picture_type              jpeg
[0:motion] [INF] [ALL] ffmpeg_output_movies      off
[0:motion] [INF] [ALL] ffmpeg_output_debug_movies off
[0:motion] [INF] [ALL] ffmpeg_bps                400000
[0:motion] [INF] [ALL] ffmpeg_variable_bitrate   0
[0:motion] [INF] [ALL] ffmpeg_video_codec        mpeg4
[0:motion] [INF] [ALL] ffmpeg_passthrough        off
[0:motion] [INF] [ALL] ffmpeg_duplicate_frames   off
[0:motion] [INF] [ALL] use_extpipe               off
[0:motion] [INF] [ALL] snapshot_interval         0
[0:motion] [INF] [ALL] locate_motion_mode        off
[0:motion] [INF] [ALL] locate_motion_style       box
[0:motion] [INF] [ALL] text_right                %Y-%m-%d\n%T
[... 144 lines cut ...]
[1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 1
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: specified frame type (3) at 0 is not compatible with keyframe interval
[1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop 1'
[1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end 1 /srv/dev-disk-by-label-Cameras/2018-07-17/10-05-19.mp4'
[1:ml1] [NTC] [ALL] mlp_actions: End of event 1
[1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 1'
[1:ml1] [INF] [EVT] event_new_video: Source FPS 10
[1:ml1] [WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted
[1:ml1] [NTC] [ENC] ffmpeg_set_codec: Using codec libx264
[1:ml1] [INF] [ENC] ffmpeg_set_quality: libx264 codec vbr/crf/bit_rate: 12
[1:ml1] [NTC] [EVT] event_newfile: File of type 8 saved to: /srv/dev-disk-by-label-Cameras/2018-07-17/10-07-19.mp4
[1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 2
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: specified frame type (3) at 0 is not compatible with keyframe interval
[1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop 1'
[... cut ...]
[1:ml1] [WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted
[1:ml1] [NTC] [ENC] ffmpeg_set_codec: Using codec libx264
[1:ml1] [INF] [ENC] ffmpeg_set_quality: libx264 codec vbr/crf/bit_rate: 12
[1:ml1] [NTC] [EVT] event_newfile: File of type 8 saved to: /srv/dev-disk-by-label-Cameras/2018-07-17/10-09-19.mp4
[1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 3
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: specified frame type (3) at 0 is not compatible with keyframe interval
`

wendy2702, 2018-07-25:
Problem solved after changing client configuration.

closed 2018-07-25 by wendy2702

ximus, 2019-01-27:
@wendy2702 howdie

can you share what you changed to fix it?

i'm getting the same sporatic cutoffs, along with:
`mjpg client timed out receiving data`

wendy2702, 2019-01-27:
Hi,

I‘m using ioBroker to visualize and control my Smarthome. To get the Camera picture on my wall mounted IPAD i select a so called Widget. Everytime the IPAD Screen is switched on the widget send a refresh command to motioneye which lead to loss of feed. After disabling the refresh it works without problem.