motioneye-project/motioneye #950: Motion Alarm Emails do not contain image attachments

zita212, 2018-07-19:
I installed yesterday the excellent motioneye on a SandyBridge Core i5, in combination with Ubuntu Server 18.04 LTS. It works great, the only issue I'm facing is that the motin alarms I'm getting do not contain attachments, are just empty emails.  I enabled the Still Image recording (as I understood is a prerequisite) but from the Pull Down menu I tried Motion Triggered and Motion Triggered - One Image, I did not try yet to record every image as a still image every second. Could this be it ?
I am using a dedicated Yahoo Email Account with TLS, when I do motion detection directly in the IP Camera I get a set of 6 attachments and it works on port 465, in motioneye port 465 did not work (Timeout error) and I had to use port 587 instead - Perhaps yahoo accepts attachments when sent via port 465 and not when sent over port 587 ? I would really love to use the motion detection in motion rather the one built in the camera as it seems more sophisticated.

ccrisan, 2018-07-19:
See https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-should-i-do-to-have-pictures-attached-to-motion-notification-emails.

JamBalaya56562, 2026-11-07:
Closing as answered with the FAQ entry linked above. In short: the email only attaches pictures that motion saved during the Attached Pictures Time Span, so Still Images must be on and that time span must be above 0.

closed 2026-11-07 by JamBalaya56562