motioneye-project/motioneye #951: motioneye service doesn't work

linoreale, 2018-07-20:
Hello, 
I've installed motioneye on a Debian 9.5 machine using the manual installation described in this page:
[](https://github.com/ccrisan/motioneye/wiki/Manual-Download-And-Installation)
and using the command 
meyectl startserver -c /etc/motioneye/motioneye.conf
it works fine.
Now I would install motioneye as a service, so I tried the instruction on point 7 here:
[](https://github.com/ccrisan/motioneye/wiki/Install-On-Debian)
but it doesn't start, and the result is this:

root@debian:/etc/motioneye# systemctl status motioneye
● motioneye.service - motionEye Server
Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Fri 2018-07-20 10:50:31 CEST; 23min ago
Process: 1487 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.c
Main PID: 1487 (code=exited, status=203/EXEC)
lug 20 10:50:31 debian systemd[1]: Started motionEye Server.
lug 20 10:50:31 debian systemd[1]: motioneye.service: Main process exited, code=exited,  status=203/EXEC
lug 20 10:50:31 debian systemd[1]: motioneye.service: Unit entered failed state.
lug 20 10:50:31 debian systemd[1]: motioneye.service: Failed with result 'exit-code'.

What could be the problem?
Thank you very much and regards

Nerdenator, 2018-09-01:
I'm experiencing a similar problem, albeit on Raspbian. 

I think the problem lies in the fact that when installed, the command `meyectl` does not exist within the directory /usr/local/bin. Instead, in my case, it's in /home/pi/.local/bin/meyectl. For some reason, instead of putting things in /usr/, Raspbian puts it in the user's home folder in /.local.

If you run `sudo find / -name meyectl`, what do you get back?

dixi83, 2018-09-02:
I think you both have not executed the command 'pip install motioneye' as root. Please try again with 'sudo pip install motioneye'

InTEGr8or, 2023-08-20:
That helped me create the missing file.

The instructions on this page, item 4, do not mention `sudo` [Install on Debian 11 (Buster)](https://github.com/motioneye-project/motioneye/wiki/Install-on-Debian-11-(Bullseye))

JamBalaya56562, 2026-11-07:
Closing as answered: `meyectl` was missing from `/usr/local/bin` because motionEye was installed without `sudo`. The current install puts it in a virtual environment instead; for anyone installing now, the steps are in the [README on `main`](https://github.com/motioneye-project/motioneye/blob/main/README.md).

closed 2026-11-07 by JamBalaya56562