motioneye-project/motioneye #954: Allow additional query strings with snapshot url.

brmo, 2018-07-23:
There are cases in which I would like to use the snapshot url and update it every xx seconds instead of streaming on devices like the RPI Zero as a Smart Mirror. However, for cache busting, you usually add another HTTP GET variable, in this case &<epoch timestamp>.  Whenever I add any query string outside of the normal "?_username=admin&_signature=" variables, i get an Unauthorized message.

ccrisan, 2018-07-23:
This is the way the signature mechanism is supposed to work. It hashes all the query arguments. Use the snapshot URL provided in the Video Streaming section and don't add anything else to it. If you must add something else to prevent caching, recompute the [signature](https://github.com/ccrisan/motioneye/blob/master/motioneye/static/js/main.js#L346) yourself.

closed 2018-07-23 by ccrisan

allmoney, 2020-07-14:
I have same problem in motioneye in Home Assistant. It's only work in same session. If I open Snapshot URL in incognito chrome mode - I see error "401: Unauthorized". Why? How fix?