motioneye-project/motioneye #955: Recorded footage freezes intermittenly

WaitWhat4355, 2018-07-23:
When pulling up the video files MotioneyeOS saves the file plays as you'd expect, but goes still frame after a few seconds for a while and _sometimes_ goes back to normal, then still frames again. 

I'm using a Raspberry Pi Zero W and it says I'm on the latest update. I started doing this after watching Linus Tech Tips' video on it, and followed their guide on their forums. I've ran it at 480P at 5FPS and 1280x720P at 10 FPS and neither has made a difference, nor has switching between H.264/OMX and Matroska Video/OMX. Is there something I'm missing, or is this a known issue? 

If needed I can provide more details. Thanks.

jasaw, 2018-07-24:
Run `ffprobe` on your recorded videos, report back what the frame rate is.
Please describe your set up, like how many cameras, FNC mode or not, using WiFi?

WaitWhat4355, 2018-07-24:
Says "1k fps" on the file. Single camera setup, wifi, no hub. I do not know what FNC mode, so I'm going to guess Fast Network Camera? If so, it is set to Off.

WaitWhat4355, 2018-07-25:
Idk what happened, seems to be working fine now,

closed 2018-07-25 by WaitWhat4355

PCcreative15, 2019-07-29:
Solution: Don't use VLC, use the Chrome browser! It is a common issue with VLC caching.

HipsteriManuli, 2020-01-03:
> Solution: Don't use VLC, use the Chrome browser! It is a common issue with VLC caching.

I have same problem and it doesn't matter will I try to watch that video with VLC, MPC-HC, Windows Mediaplayer or with Chrome.  The problem is on those video files and not the player.

jasaw, 2020-01-06:
@HipsteriManuli If your hardware is not capable of encoding and saving the video in real-time, you will get intermittent pauses in the recorded video. Try lowering resolution, frame rate, bit rate. Try using hardware accelerated encoder.