motioneye-project/motioneye #957: Is full access to Dropbox necessary?

multiformeingegno, 2018-07-24:
I am always a bit worried when I have to give full access to my Dropbox account...

ccrisan, 2018-07-27:
@multiformeingegno honestly I have no idea. [Here](https://github.com/ccrisan/motioneye/blob/master/motioneye/uploadservices.py#L436) you can find the Dropbox upload service implementation. If you can spot a place where this behaviour can be changed, I'm open to suggestions.

JamBalaya56562, 2026-11-07:
Closing, as no change was suggested after ccrisan's question above. Which access a Dropbox app asks for, a single app folder or the whole Dropbox, is chosen when the app is registered with Dropbox, not in motionEye's code, so changing it would need a new app registration. Please open a new issue if you would like to propose that.

closed 2026-11-07 by JamBalaya56562