motioneye-project/motioneye #959: permissions error

Olegone23, 2018-07-25:
Hi Ccrisan,

Could you pls. help with the issue below.
Seems some permissions are wrong I can't apply any video cam settings, getting errors below while pushing Apply button.
Also can't see motioneye log file in /var/log  seems not created (motion.log is ok)

9f5186ac7995dcc2186c9606 (127.0.0.1) 0.69ms
   ERROR: 'manual_snapshots'
Traceback (most recent call last):
  File "/home/developer/.local/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/home/developer/.local/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
    result = method(self, *args, **kwargs)
  File "/home/developer/.local/lib/python2.7/site-packages/motioneye/handlers.py", line 292, in post
    self.set_config(camera_id)
  File "/home/developer/.local/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
    return func(self, *args, **kwargs)
  File "/home/developer/.local/lib/python2.7/site-packages/motioneye/handlers.py", line 527, in set_config
    set_camera_config(int(key), cfg, check_finished)
  File "/home/developer/.local/lib/python2.7/site-packages/motioneye/handlers.py", line 374, in set_camera_config
    local_config = config.motion_camera_ui_to_dict(ui_config, local_config)
  File "/home/developer/.local/lib/python2.7/site-packages/motioneye/config.py", line 781, in motion_camera_ui_to_dict
    '@manual_snapshots': ui['manual_snapshots'],
KeyError: 'manual_snapshots'
   ERROR: 500 POST /config/0/set/?_=1532529346807&_username=admin&_signature=86e8ec2d8315d3dd9c1b386201dad1043d9338d7 (127.0.0.1) 2.28ms

Kind Regards,
Oleg

ccrisan, 2018-07-25:
Please remove and add back your cameras. You probably have upgraded motion or motionEye and there seems to be some kind of configuration incompatibility.

ccrisan, 2018-07-27:
@Olegone23 what versions (motionEye, motion, OS, hardware) do you use?

ccrisan, 2018-07-27:
Then why do you run motionEye 0.35.2? Please make sure you run the latest version by using the correct python interpreter/virtualenv.

Olegone23, 2018-07-27:
It is just copied new version 0.39.2 to old motioneye-0.35.2-py2.7.egg folder

ccrisan, 2018-07-27:
@Olegone23 I don't get it. What's copied where? What version do you see in the UI?

Olegone23, 2018-07-27:
0.39.2

Olegone23, 2018-07-27:
Firstly it was 0.35, but i've just updated it to version 0.39.2 (put all files of 0.39.2 to the 0.35 folder).

ccrisan, 2018-07-27:
Did you manually copy the new files into the old folder? If yes, please never do that. Uninstall it with pip, see if you can cleanup everything there and install/upgrade it with pip as indicated on the wiki page.

Olegone23, 2018-07-27:
Sorry for my wrong explanation, I have no pip (installed it from source), but if u think something could mixed, old with new, I will try to reinstall it. What is a minimum permissions should be applied while installing all components?
And why motioneye need ffmpeg? I i precompiled motion with ffmpeg, always though, motion needs ffmpeg only.

Olegone23, 2018-07-27:
And also may be u know what is a best configuration of motion not to freeze while working with a lot of full hd cams? I can see that motion is somehow is a bottleneck, I use a vlc embed instead of img tag in main screen of motioneye, and still freezes happens very often and broken images

Olegone23, 2018-07-27:
Or I come to a conclusion that it is a bad idea to detect motion from many cams using one multicores processor inside the PC. Many modern cams do it easely by themself, so the best way just to get the motion trigger from them? What do u think?

ccrisan, 2018-07-27:
Please see our wiki, particularly [this](https://github.com/ccrisan/motioneye/wiki/Installation) article.

You can do the motion detection on the central server if you have enough CPU power. Otherwise, you're free to experiment various scenarios.

closed 2018-07-27 by ccrisan