motioneye-project/motioneye #960: Use GPU on Odroid XU4

kevkid, 2018-07-26:
I am using Ubuntu 16.04 but got latest Bionic sources and upgraded packages that way:
```
deb http://ports.ubuntu.com/ubuntu-ports/ bionic main universe restricted
deb-src http://ports.ubuntu.com/ubuntu-ports/ bionic main universe restricted

#deb http://ports.ubuntu.com/ubuntu-ports/ bionic-updates main universe restric$
#deb-src http://ports.ubuntu.com/ubuntu-ports/ bionic-updates main universe res$

deb http://ports.ubuntu.com/ubuntu-ports/ bionic-backports main restricted
deb-src http://ports.ubuntu.com/ubuntu-ports/ bionic-backports main restricted

deb http://ports.ubuntu.com/ubuntu-ports/ bionic-security main restricted
deb-src http://ports.ubuntu.com/ubuntu-ports/ bionic-security main restricted
deb http://ports.ubuntu.com/ubuntu-ports/ bionic-security universe
deb-src http://ports.ubuntu.com/ubuntu-ports/ bionic-security universe     
deb http://ports.ubuntu.com/ubuntu-ports/ bionic-security multiverse          
deb-src http://ports.ubuntu.com/ubuntu-ports/ bionic-security multiverse
```
Here is what I tried
- [x] Updating to latest motion eye (0.39.2)

- [x] updating to latest motion (4.1.1) using: [https://launchpad.net/ubuntu/+source/motion](https://launchpad.net/ubuntu/+source/motion) where now I finally see H.264/OMX

- [x] Selecting H.264/OMX

I still get high cpu utilization (according to top the motion process is at 110%+) and when I try to look at gpu utilization I get:
```
$ cat /sys/devices/11800000.mali/utilization
0
```
Could I be missing something?

I have 2 cameras which are in fast network mode they are raspberry pi zeros. It is being recorded at 1024x768@20fps. Image quality is at 5% and it is detecting motion.

jasaw, 2018-07-26:
I have blacklisted `h264_omx` encoder in motion software because of a lock up issue on Raspberry Pi. See https://github.com/ccrisan/motioneye/issues/930
You'll need to comment out `h264_omx` from the blacklist and recompile motion software.

TopperBG, 2018-07-26:
@kevkid how did you manage to isntall from PPA  https://launchpad.net/ubuntu/+source/motion ?

kevkid, 2018-07-26:
@TopperBG I installed it from the deb file located: https://launchpad.net/ubuntu/+archive/primary/+files/motion_4.1.1-1.1build1_armhf.deb

It is under the cosmic cuttlefish branch.

xu4user, 2018-10-16:
Can you tell me how to install the motion_4.1.1-1.1build1_armhf.deb by doing the dpkg -i then followed by a apt-get install -f? Do you have to recompile only the motion software or the ffmpeg also? Do you recompile the motion software with the pip install motion command?

kevkid, 2018-10-16:
@xu4user did you try `dpkg -i`?

xu4user, 2018-10-16:
No, that is what I was asking. What are the procedures for doing this... so it would be dpkg -i motion_4.1.1-1.1build1_armhf.deb, then follow that by a apt-get install -f... correct? After that, then I comment out the h264_omx line in the motioneye.conf file and then do a pip2 install motioneye?? Is this correct?

jasaw, 2019-07-01:
For XU4 hardware video encoding, you need to use `h264_v4l2m2m` encoder. Make sure your ffmpeg supports h264_v4l2m2m, then change `movie_codec` parameter in the `camera-1.conf` file to select the `h264_v4l2m2m` encoder to this: `movie_codec mp4:h264_v4l2m2m`.

JamBalaya56562, 2026-11-07:
Closing as answered: as jasaw wrote above, the XU4 encodes in hardware with `h264_v4l2m2m`. motionEye now picks that codec by default when Motion and ffmpeg support it and `h264_omx` is not available.

closed 2026-11-07 by JamBalaya56562