motioneye-project/motioneye #961: Unable to open video device

RedGuilhas, 2018-07-27:
Hello,

I have a Xiaofang camera that I flashed the Fang Hacks firmware. It works by rtsp. I can view it through VLC so I know it works.
When I added it to MotionEye I got a blank screen with the message "Unable to open video device".

This all I have in the logs:

jul 27 17:28:40 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:28:40 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:28:40 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:28:50 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:28:50 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:28:50 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:28:50 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:29:00 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:29:00 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:29:00 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:29:00 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:29:10 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:29:10 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:29:10 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:29:10 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:29:20 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:29:20 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:29:20 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:29:20 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:29:30 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:29:30 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:29:30 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:29:30 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:29:40 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:29:40 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:29:40 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:29:40 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:29:50 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:29:50 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:29:50 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:29:50 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:30:00 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:30:00 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:30:00 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:30:00 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:30:10 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:30:10 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:30:10 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:30:10 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:30:20 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:30:20 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:30:20 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:30:20 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:30:30 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:30:30 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:30:30 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:30:30 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:30:40 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:30:40 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:30:40 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:30:40 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:30:50 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:30:50 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:30:50 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:30:50 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
jul 27 17:31:00 hassi motion[1686]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:31:00 hassi motion[1686]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/unicode/)
jul 27 17:31:00 hassi motion[1686]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:31:00 hassi motion[1686]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/unicode/): Server returned 404 Not Found
[... cut ...]
jul 27 17:38:20 hassi motion[6010]: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
jul 27 17:38:20 hassi motion[6010]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.78/)
jul 27 17:38:20 hassi motion[6010]: [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 404 Stream Not Found
jul 27 17:38:20 hassi motion[6010]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.78:554/): Server returned 404 Not Found

I have Motioneye installed on a old netbook running Ubuntu.

ccrisan, 2018-07-27:
If you think your URL, protocol, credentials & everything are correct, please [Report A Motion Issue](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue).

closed 2018-07-27 by ccrisan

ageorgios, 2018-08-28:
Same here, did you find a way to use Xiaofang with fang-hacks to motioneye?

RedGuilhas, 2018-08-28:
Yes you have to put the link correctly.

rtsp://cameraip/unicast(without the forward slash in the end)

ageorgios, 2018-08-28:
Is it stable for you? 
It is somewhat stable after removing audio from snx_rtsp server that is running in xiaofang (removing the -a option)
sometimes it crashes the rtsp server on the xiaofang camera