motioneye-project/motioneye #964: Number of locally saved images do not match the number of FTP images

discarn8, 2018-07-30:
Hello All:

I'm using the Upload Service beneath the File Storage section, in order to also save a copy of my motioned activated images to an FTP server (as well as the local SD card).  

My issue is: when I view the images that were saved on the FTP server (via PHP browsing script), the number of images that have been saved to FTP, do not match the number of images on the local motioneyeos SD card.  The SD card contains several motion activated images that are not on my FTP server. Initially, I suspected my WiFi just couldn't keep up with the demand, so I switched to ethernet.  No change.  
Then, I suspected my Pi2 couldn't handle the load and I switched to a Pi3 - no change.  I'm connecting the Pi to a gigiabit port on my wifi router which is connected to a gigabit switch back out to a gigabit port on my FTP server.  Network bandwidth used appears to be negligible.  Has anyone else experienced this and if so, what was your fix?

Thanks in advance
Rich

discarn8, 2018-10-19:
Hello All:

I'm using the Upload Service beneath the File Storage section, in order to also save a copy of my motioned activated images to an FTP server (as well as the local SD card).

My issue is: when I view the images that were saved on the FTP server (via PHP browsing script), the number of images that have been saved to FTP, do not match the number of images on the local motioneyeos SD card. The SD card contains several motion activated images that are not on my FTP server. Initially, I suspected my WiFi just couldn't keep up with the demand, so I switched to ethernet. No change.
Then, I suspected my Pi2 couldn't handle the load and I switched to a Pi3 - no change. I'm connecting the Pi to a gigiabit port on my wifi router which is connected to a gigabit switch back out to a gigabit port on my FTP server. Network bandwidth used appears to be negligible. Has anyone else experienced this and if so, what was your fix?

Thanks in advance
Rich

JamBalaya56562, 2026-11-08:
Closing, as you were going to try jasaw's suggestion. As he said, an upload that fails is not retried, and that is still the case: only an expired login is retried, so a file whose FTP transfer fails stays local only, with the error in `motioneye.log`. Syncing the media folder with rsync, as jasaw does, avoids that. A retry for failed uploads would be a feature request of its own.

closed 2026-11-08 by JamBalaya56562