motioneye-project/motioneye #965: Camera Resolution and Video rotation ignored.

surak, 2018-07-30:
motionEye Version | 0.39.2
Motion Version | 4.1.1
OS Version | Raspbian 9.4
 
I installed it on friday, and everything was fine. Today, the image was inverted and in lo-res (see pic).

[image]

ccrisan, 2018-07-30:
@surak can we see a screenshot of your preferences (the first part of the settings menu)?

surak, 2018-07-30:
In fact, it's more being ignored. The text overlay is also ignored. 
[image]

ccrisan, 2018-07-30:
Your prefs are fine. There might be an issue with the camera driver, saying it supports higher resolutions whereas the camera doesn't actually support it. Try lowering the resolution and see if you can make it obey what you're asking it for.

Regarding the text overlay, why do you think it's ignoring it?

surak, 2018-07-31:
The camera WAS capturing at the right resolution with motioneye - it's the official Pi NoIR I wonder what happened after I updated everything. If I take pictures with raspistill they are at full resolution.

If you look at this picture, there is the camera name on the left, which is set up there. And the rotation is 180º, as the camera is upside down. After the update, it's upside down and the label is default, as you can see on the previous images.

[image]

ccrisan, 2018-07-31:
Beats me. Try readding it from scratch. Try adding it as a MMAL camera, if say it's the official RPi camera.

surak, 2018-07-31:
I have no clue what's going on either. I did all of it (I wouldn't ever open an issue before doing something). I only use it under raspbian because this device specifically has a screen - in all others, I use motioneyeos and it works like a charm. 

Will clear up raspbian and start clean.

surak, 2018-07-31:
Now with 0.39.3 it's all back to work

closed 2018-07-31 by surak