motioneye-project/motioneye #966: Running MotionEye in Docker Container

TexanScot, 2018-08-02:
I'm working on a system that will have multiple (3 or 4) Pi Zero camera's each of which will run under MotionEyeOS. I'm intending to stream output from these cameras to a central hub that will handle motion detection, notification, etc. 

I'd like to use spare capacity on an existing iMac to run the central MotionEye hub and thought of using Docker to accomplish this. I've confirmed that I can successfully connect to a containerized version of Docker on port 8765 of the host machine. 

I'd like to use push notifications, using Pushover, to notify me of movement detection. I understand this can be done by calling the Pushover API from Python. My question is whether the containerized version of Docker has the necessary code (Python) available for this to work?

kleini, 2018-08-02:
Within the Docker container motionEye is installed as within a normal system. The Docker container is just a minimized version of a normal system containing only those parts, that are necessary to run the single process motionEye.

I think with "Pushover" you mean this piece of software https://pushover.net/ ?
Unfortunately I did not do anything yet with the motion notifications of motionEye. Having a look into the administration interface of motionEye show there some method, that allows you to POST some json, but I have no idea, how to define the JSON that can be posted there with.

TexanScot, 2018-08-02:
Yes, your link refers to the software that I would like to use for push notifications.

There are a number of sites that demonstrate the python code necessary to construct and send the necessary POST to Pushover, e.g. [https://www.raspberrycoulis.co.uk/coding/add-push-notifications-motioneye-os/](url). MotionEye is then configured to run this Python script when motion is detected.

I looked at the Build file for the MotionEye Docker container and I see a number of references to Python in the RUN section of the file. So I suspect that the code necessary to run Python within the container may already be in place. Guess I will just have to give it a go and see what happens!