motioneye-project/motioneye #967: [NOT ISSUE] Trigger video / image saving manually [NOT ISSUE]

TheRaspyBoi, 2018-08-03:
Is it possible to manually activate the video / image saving?
Basically, the idea is to manually activate the script for saving video / image when a motion is detected.

Sam12604, 2018-08-07:
Assuming that you have installed MotionEye on Raspian and you want to be able to record video and pictures by pressing a button you just have to change the settings. For manual pictures you have to go to the control panel, log in as admin and go to Still Images; you have to make sure advanced options are enabled. You just have to press Enable Manual Snapshots and apply settings. 
[image]
You should get this button:
[image]
However, on the Movies tab there is no option for manual button, at least for me, so you have to use the config file on the pi, go to the MotionEye folder which should be located at /etc/motioneye
[image]
Open thread-1.conf (for camera 1, thread-2.conf if you have another camera) scroll until you see 'manual_record' and change off to on:
[image]
Go back to the control panel, now there should be a symbol like this to record:
[image]

Sam12604, 2018-08-07:
Oh sorry about that that was my mistake, not sure how you could do that but I will have a look for you. :smile:

TheRaspyBoi, 2018-08-07:
No problem, thanks anyway ;)

JamBalaya56562, 2026-11-09:
This is possible since #2691 (0.45.0b1): with "Manual motion event button" enabled in the motion detection settings, the camera gets a button to start and stop a motion event, and a logged-in request can do the same with `POST /action/<camera_id>/eventstart` and `/eventend`. Closing as done.

closed 2026-11-09 by JamBalaya56562