motioneye-project/motioneye #970: Connecting MotioneyeOS Camera to Motioneye Hub

TexanScot, 2018-08-06:
I have successfully configured a Pi Zero camera using MotionEyesOS. I can access the camera by entering the IP address of the Pi Zero in my browser and Motioneye identifies the device as a V4L2 camera running on device /dev/video0.

Since I intend to install a number (3 or 4) Pi Zero cameras my intent was to designate a central Motioneye hub that would be able to access all of the cameras. My idea was to use my Mac Media server running MotionEye in a Docker container as the hub, since that has the processing power and storage available. I am able to successfully run MotionEye in the container and can connect to it by accessing /localhost:8765 on my browser.

I see the MotionEye "Preferences" and "General Settings" menus on the left hand side of the screen and the message "You have not configured any camera yet....." on the right hand side. However when I click on this message and attempt to add a camera (Remote MotionEye Camera)  I am unable to enter the address for the Pi Zero Camera successfully (camera box is greyed out?).

I was following this link [https://randomnerdtutorials.com/cctv-raspberry-pi-based-system-storage-motioneyeos/] for guidance and I note that it advises setting up a local camera on the hub for initial configuration purposes. Obviously I can't do this since the hub is running in a docker container.

Is there something that I'm doing incorrectly or it is simply not possible to utilize the configuration that I have proposed? My intent was that the Pi Zero cameras would eventually run as "Fast Network Cameras" and the hub would execute motion detection and provide storage for pictures and videos. If I switch on the Fast Network Camera option on the Pi Zero would I be able to add it on the hub?

Appreciate your feedback on my proposed configuration and how to make it work.

TexanScot, 2018-08-06:
OK.......I managed to get my desired configuration working by specifying the Pi Zero camera as a Fast Network Camera and using the streaming URL with the MotionEye container.

Pictures and video are being successfully stored on one of the media drives installed on my Mac server.

There is one thing that I haven't figured out however and that is that the timestamp on the MotionEye hub is incorrect. In MotionEyeOS this can be set in the General Settings area however it seems to be missing from MotionEye. MotionEye is version 0.39.3, Motion 4.1 running on Ubuntu 17.10.

Where can I set the timestamp in Motion. I specified a folder for /etc/localtime when starting the container however as far as I can tell this folder is empty. Is there meant to be a file here that tells MotionEye the current time?

Thanks for a great piece of software.

TexanScot, 2018-08-07:
Based on some further reading I created a file /motioneye/localtime/timezone that I pass into the container as /etc/localtime:ro. The timezone file contains only the string "US/Chicago" which I think ought to define the host timezone correctly.

However this doesn't seem to make any difference and I still see the incorrect time in Motioneye. Can anyone advise what needs to be passed to the container as /etc/localtime

ccrisan, 2018-08-21:
The original issue was about failing to add a camera. Please attach a screenshot to exemplify your problem.

Regarding your second commend, motionEye should not be used to control the timezone unless used in conjunction with motionEyeOS.

TexanScot, 2018-08-21:
OK, I am going to try to provide a detailed description of what I have configured and tried to make work.

I'm working on a Mac with the latest version of docker and Kitematic. On the Mac I have created empty folders (config, media and var) under a ~/Docker/MotionEye folder.  On Kitematic I download the AMD64-Master version of MotionEye and configure it as follows (see attached screenshots):

Ports:     8765:8765
               8081:8081

Files:   /var/lib/motioneye ------> ~/motioneye/media
            /etc/motioneye ------> ~/motioneye/config
           /var/run/motion -------> ~/motioneye/var

Also see the screenshots of the initial MotionEye screen when I connect to localhost:8765 and the message displayed when I attempt to add a camera.

Can you please clarify your comment regarding MotionEye and the timezone? Do you mean that the correct time can only be acquired when running MotionEye under MotionEyeOS and that there is no way of setting the correct time in MotionEye when running under Docker?

[image]

[image]
[image]
[image]

[image]

[image]

TexanScot, 2018-08-27:
Hoping to get some feedback regarding my post above. At present I'm unable to run Motioneye under Docker.

JamBalaya56562, 2026-11-08:
Closing, as you got the camera working on the hub as a Fast Network Camera with its streaming URL, and this was with the Docker image of 2018 under Kitematic. For the time on the hub, current images take the time zone from `-e TZ=US/Chicago` or a mounted `/etc/localtime`, see [docker/README.md](https://github.com/motioneye-project/motioneye/blob/dev/docker/README.md). Please open a new issue if a current image does not start on your Mac.

closed 2026-11-08 by JamBalaya56562