motioneye-project/motioneye #972: When will there be an API?

doubledrat, 2018-08-09:
looked at the API pages and they were last updated 18 months ago.  Any idea when the API is likely to be implemented?  This is essential for integrating motioneye with smart home implementations...

thanks for a great system!

kolossboss, 2018-10-17:
I would like to see this too!

roger-zacharczyk, 2019-01-22:
@ccrisan two questions for you:
1. Do you have any branch with started API implementation I could start with and send you some pull requests? I'm willing to help with implementing the API.
2. I've seen the draft. Could we add action buttons there?

ccrisan, 2019-01-22:
@roger-zacharczyk no plans to add an API anytime soon. I'm busy with many other projects. I will accept PRs though, if anyone is willing to actually take care of this big task and stick around for a while...

roger-zacharczyk, 2019-01-23:
@ccrisan I assume the authentication part (https://github.com/ccrisan/motioneye/wiki/API-(Draft)#authentication) is ready?

So, I can proceed with extending the handler_mapping in server.py. 

And refactoring existing code. Because for example API should return the list of cameras and existing UI needs it too but now it uses /config/list end point instead of /api/cameras.