motioneye-project/motioneye #973: MotionEye Smart Mask set to 10 disables Mask

AZRadiohead, 2018-08-12:
motionEye Version | 0.39.3
Motion Version | 4.0 (4.0.1 reported by APT when trying to upgrade motion)
OS Version | Ubuntu 17.10

When setting Mask to On, Mask Type Smart, Smart Mask Sluggishness 10 and clicking Apply the Mask turns to Off.  All other settings, 1-9 work properly.  This was tested on multiple camera feeds and after several restarts with the same result.

ccrisan, 2018-08-13:
The two types of mask can't be used at the same time.

AZRadiohead, 2018-08-14:
The issue is that when using SMART MASK and adjusting it's setting, "Smart Mask Sluggishness", a setting of 10 turns Off the mask.  "Smart Mask Sluggishness" has an option of 1 through 10 and 10 causes the Mask ON/OFF setting to reset to Off.  Can you fix this?  I'm not trying to use two masks at once.  It's a broken setting.

lutierigb, 2020-01-21:
@ccrisan  please re-open. this is an issue. I had the same experience.

ccrisan, 2020-01-25:
It was indeed a bug. Should be fixed in https://github.com/ccrisan/motioneye/commit/c955e03798aeea755801c0bf19abcb91a21e24a4.

closed 2020-01-25 by ccrisan