motioneye-project/motioneye #976: MotionEye on raspbian: no camera detected

nakiot, 2018-08-16:
Hello everybody,
I have a problem with my camera on raspbian through MotionEye.
I got raspberry pi 3 b+ and a camera night vision bought on the website modmypi.
I installed MotionEyeOs and everything was working properly. 
I installed the latest version of raspbian
I followed this topic with the latest version of everything: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
And now when I go to 127.0.0.1:8765, my camera is not detected.
With the command "vcgencmd get_camera": supported=0 detected=0
Do you have any idea what can I check?
Thanks a lot!

Nerdenator, 2018-09-01:
Have you enabled the camera in raspi-config under interfaces?

JamBalaya56562, 2026-11-08:
Closing, as this got no further replies. `vcgencmd get_camera` giving `supported=0 detected=0` means the camera interface was not enabled in the OS, as nerdenator suggested with raspi-config, so motionEye had no camera to list. That was the legacy camera stack; on current Raspberry Pi OS the camera goes through libcamera, see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack).

closed 2026-11-08 by JamBalaya56562