motioneye-project/motioneye #977: Layout columns not working on RPI 7" Touchscreen?

RobertKrens, 2018-08-16:
Hello, 

First of all I'm really happy with the MotionEyeOS. It works great so far!

Setup: 
RPI3B - Wifi connected - Camera connected =  Server
RPI Zero - Wifi connected - Camera connected = Client
RPI3B+ - Wifi Connected - 7" RPI Touchscreen connected = Wall display

I've added the RPI Zero camera in the RPI3B configuration. This works flawless. 
Then the RP3B+ With touchscreen boots up with Chromium going to the RP3B Servers IP.

I want to have both camera's been shown next to eachother on the 7" Touchscreen. At this point they are shown underneath eachoterh and I need to scroll down to see the other frame. I've been playing arround with resolutions but that doesn't alter the size of the frames. Also been chaning the "fit frames vertically" and "Layout rows" and both won't change anything. While opening the RPI3B MotionEye Server on my laptop I can alter the Layout rows and it alters the frames. But the changes won't have any affect on the RPI 7" touchscreen.

Is there anyone who can point me in the right direction?

Thanks in advance and best Regards, 

Robert Krens

ccrisan, 2018-08-16:
The screen is identified as a "small" screen - assimilated to a mobile device. On mobile devices, frames are never laid out on columns. A workaround would be to zoom out a bit (using Ctrl -) in Chrome.

RobertKrens, 2018-08-16:
Dear Calin, 

That's a great workarround! Thanks!

Regards Robert

closed 2018-08-16 by RobertKrens