motioneye-project/motioneye #978: feature request:Prevent disk filling

doubledrat, 2018-08-17:
I know 3 years ago you said you had no intention of adding this feature, but it's impossible to know how much space will be consumed in advance.  Surely an option to maintain at least x% of free space on the output volume is a sensible option to have as well as keeping files for a certain amount of time?

it would just have to run before and/or after movies or pictures are written and remove oldest files until the threshold has been reached.

the alternative of having nothing written because the disk has gone full is rather sub-optimal.

ccrisan, 2018-08-17:
It's one of the first features to be added in the next version.

JamBalaya56562, 2026-11-09:
Closing as a duplicate of #302, which asks for the same. It was planned back then but has not been done yet; please follow #302 for it.

closed 2026-11-09 by JamBalaya56562