motioneye-project/motioneye #981: Can't Add a Camera

TexanScot, 2018-08-20:
I'm attempting to run MotionEye in a docker container on my Mac using Kitematic. The intent is to have MotionEye receive feed from a MJPEG camera that is operating on a Rasperry Pi Zero. 

The container starts and I can connect to the server on localhost:8765. I see the MotionEye screen with the message "You have not configured any camera yet. Click here to add one..."

However when I click on this I get a pop-up dialog that states "Please apply the modified settings first!". When I then click OK nothing happens.

I have:
/var/lib/motioneye       ------>  ~/motioneye/media
/etc/motioneye           ------>  ~/motioneye/config
/var/run/motion          -------> ~/motioneye/var

Appreciate feedback regarding where I am going wrong with the configuration.

mhaddy, 2018-12-28:
@TexanScot did you get this issue figured out? I am having the same issue.

motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | Raspbian 9.4

JohanKlos, 2019-01-02:
I'm having the same issue on my old raspberry pi 2 (which is running osmc, which could have something to do with this).

motionEye Version | 0.39.3
Motion Version | 4.2.1
OS Version | Linux 4.14.78-4-osmc

mhaddy, 2019-01-02:
My issue was solved with a Pi restart.

closed 2019-01-11 by ccrisan

DelightedD0D, 2019-02-19:
Getting the same thing "Please apply the modified settings first!", rebooted and re-flashed several times, but I keep getting this error even though no settings have been modified :(

TexanScot, 2019-02-19:
I was never able to resolve the issue and eventually gave up. I may try again when I have the need to integrate a few more cameras into my system.

DelightedD0D, 2019-02-19:
@TexanScot I was able to get past this by setting a breakpoint in the javascript where that warning was being thrown and using the console to manually edit the vars to remove the supposed "pending changes"

After that, it let me in fine. 

There seems to be an issue in the javascript that sets up the "options" panel when you log in. It seems like sometimes when the values are set in those fields, the app thinks that you have made changes to them and therefore wants you to submit them before doing anything else.

If it happens to me again, ill debug deeper down the rabbit hole.

miwallac, 2019-02-20:
This was resolved for me by cleaning up a stray, empty media folder at /var/lib/motioneye.  YMMV.

zevele, 2019-03-15:
I've managed to overcome this issue with the following (don't know which one did the trick):
1. disable all passwords 
2. disable advanced settings
3. reboot.

Then when clicking the "You have not configured any camera yet. Click here to add one..." link I was able to configure the camera.